dreemurrs-embedded/Pine64-Arch #1: accessing the phone...?

Tarcaxoxide, 2020-03-11:
i am not sure what type of connection i should setup on the host side.
and
the ip address doesn't seem to be accessible?

i tried
(telnet and ssh) 172.16.42.2
both just hang and then timeout
i also noticed that the hostname is 'alarm' and so i tried 
ping alarm
but the name or servcie is not known

Tarcaxoxide, 2020-03-11:
perhaps you could add a user and run a bash script on launch?
i'd do it myself but i don't know how to setup auto-login

`#!/bin/bash
while :
do
    dhcpcd 
    ifconfig
   sleep 5
done
`

Danct12, 2020-03-11:
The device IP address is `172.16.42.1`.
The user and password is alarm/alarm.

Tarcaxoxide, 2020-03-14:
> The device IP address is `172.16.42.1`.
> The user and password is alarm/alarm.
> 
> If that doesn't work, make sure you're on the latest rootfs: https://github.com/dreemurrs-embedded/Pine64-Arch/releases/tag/20200308

im on the latest . does not work.
is there a specific network type i need to use?
right now i just have it setup with a Ethernet type connection
ipv6 'Link-Local Only'
ipv4 'Shared to other computers'

Tarcaxoxide, 2020-03-14:
> The device IP address is `172.16.42.1`.
> The user and password is alarm/alarm.
> 
> If that doesn't work, make sure you're on the latest rootfs: https://github.com/dreemurrs-embedded/Pine64-Arch/releases/tag/20200308

i also have another machine connected to this system.
its ip is being reported as 10.42.1.105

Tarcaxoxide, 2020-03-15:
> The device IP address is `172.16.42.1`.
> The user and password is alarm/alarm.
> 
> If that doesn't work, make sure you're on the latest rootfs: https://github.com/dreemurrs-embedded/Pine64-Arch/releases/tag/20200308

i tried to also
ping alarm
but that didn't seem to work.
is hostname not setup?

asriel-danctnix, 2020-03-17:
You'll need to manually assign an IP to the interface as DHCP is currently broken: `172.16.42.2/24`

An example would be this:
`sudo ip addr add 172.16.42.2/24 dev enp0s20f0u14`

After that, you should be able to connect to the device through `ssh alarm@172.16.42.1`

Tarcaxoxide, 2020-03-17:
> You'll need to manually assign an IP to the interface as DHCP is currently broken: `172.16.42.2/24`
> 
> An example would be this:
> `sudo ip addr add 172.16.42.2/24 dev enp0s20f0u14`
> 
> After that, you should be able to connect to the device through `ssh alarm@172.16.42.1`

that worked , thanks 

> You'll need to manually assign an IP to the interface as DHCP is currently broken: `172.16.42.2/24`
> 
> An example would be this:
> `sudo ip addr add 172.16.42.2/24 dev enp0s20f0u14`
> 
> After that, you should be able to connect to the device through `ssh alarm@172.16.42.1`

that worked thanks :)

closed 2020-03-17 by Tarcaxoxide