dreemurrs-embedded/Pine64-Arch #2: pine64 corrupted .db file

ejiek, 2020-04-11:
Due to `pine64` being defined as `http` repo, my `/var/lib/pacman/sync/pine64.db` got populated by some html page (cellular provider redirect or https redirect in the repo itself).
It resulted in following error during pacman update:
```
error: could not open file /var/lib/pacman/sync/pine64.db: Unrecognized archive format
error: failed to prepare transaction (invalid or corrupted database)
```

I had to modify `http` to `https` for `pine64` repo in `/etc/pacman.conf` and delete `/var/lib/pacman/sync/pine64.db`.
After this updates started to work like a charm.

Second thing is SigLevel which set to `Never`.
I don't think it's a good combination with `http`.

I haven't found where to patch it in this repo so I ask you to update it for the next release if you don't mind =]

Danct12, 2020-04-11:
I'll think of signing the packages in next release, right now we're preparing to push Phosh UI packages.

Danct12, 2020-06-22:
I've solved the https issue with the latest installations.

Still hasn't signed packages yet.

Danct12, 2020-06-29:
Issue solved with the latest image.

closed 2020-06-29 by Danct12

kescherCode, 2021-07-11:
Hmmm... I found that the current images have "SigLevel = Never" in some of the repos, and that the packages don't actually have signatures (.sig files).
So the packages aren't signed.