dreemurrs-embedded/Pine64-Arch #3: Image Upgrade

crwagn, 2020-06-25:
I attempted to do a pacman -Syu ran into the pine64 repo issue. I fixed the pine64 repo issue. However, it keeps trying to pull a uboot firmware update dated for April(202004) that isn't up on the site, rather than seeing and getting the newly updated uboot firmware dated for July(202007). Even if I just pull the July one, it says there are no signatures, so it wont update.

Not sure how to make that stop, but I was hoping to update it and get phosh on rather than waiting for the new image to be dropped.

Thanks.

Danct12, 2020-06-26:
U-Boot 2020.04 is intended, it contains support for crust.

crwagn, 2020-06-26:
So where would that be located? Because it is trying to get it from https://p64.arikawa-hi.me/pine64/ but that doesnt have any of your files anymore, so to update I fixed it to look at https://p64.arikawa-hi.me/pine64/aarch64/ 

However, the only uboot firmware there is the 2020.07 rc one, so where is the 2020.04 one, that it seems to be seeing but cant download, because I dont see where to point it to get the update to be happy for most of the files other than https://p64.arikawa-hi.me/pine64/aarch64/ and I dont see the 2020.04 firmware anywhere on the site.

Thanks for the fast response.

Danct12, 2020-06-26:
Regarding the URL change, that was intended to split between phosh and the main repo so maintaining it would be much easier for me.

About the U-Boot firmware, I just reuploaded the correct package, let me know if it works.

crwagn, 2020-06-27:
Sorry for the slow delay, with a the dns and https fix, I got the update to go through. Thank you! Should I just be pulling all the phosh8 stuff from the phosh section of your site now?

crwagn, 2020-06-27:
Actually, now that I've reboot after the install, I am only getting the "white" led after the green one and it's staying on with no screen coming up. Even sitting there for a while, nothing changes until i hard shutdown the phone, even pushing the power button doesn't do the power down it did before.

crwagn, 2020-07-02:
Thanks for the phosh push, but I have a quick question and just tell me if I should just open a new issue or not.

Did you push a memory speed increase with the phosh update? 

I'm getting random crashes and freezes that remind me of postmarket after their memory speed up, before they reverted it. I found out I have some of the worst memory capability possible, I cant even get 512 to work stably for long periods of time.

Danct12, 2020-07-02:
Yep, I increased it to 624mhz and it seems to work fine on both my v1.0 and v1.2a.

About the random crashes, which is the stable?

crwagn, 2020-07-05:
I have the 1.1 and have terrible crashes with all updated speeds. There was testing going on in the pine64 forums to find out about usable memory speeds to control the crashes. 

Though, the crashes with your arch are not as bad as they were on pmOS, I never got to stable speeds with any of the original test speeds all the way down to 512. However, for the few of us that never got stable, a new version was released with 492MHz speed and it seemed to work for us. 

Some of the unlucky braveheart users have hardware that cant run any level of up speed for the ram and it causes random crashes that lock up the whole system. Sometimes it does a soft crash and restart. But sometimes, it just freezes completely until you hard power off the device as well.

Here's a link to the forum post I made on my freezes in pmOS: https://forum.pine64.org/showthread.php?tid=9891

And this is the post of the guy testing memory speeds, where in some of the later pages he adds a 492 speed for people like me and a few others who cant run anything above base due to losing the silicon lottery: https://forum.pine64.org/showthread.php?tid=9832

crwagn, 2020-07-05:
Thanks for opening a new issue, I'll go ahead and close this one now!

closed 2020-07-05 by crwagn