dreemurrs-embedded/Pine64-Arch #5: Call audio switching

nickgirga, 2020-07-01:
Audio profiles do not automatically switch when sending or recieving calls. I had to go into settings and change "System Settings > Sound > Output > Configuration" to "Make a phone call" and the Output Device option next to it to "Internal Earpiece" while in the middle of a call. Then I had to switch it back after I hung up. If the switch is not done, absolutely no audio is transmitted either way, so you'd probably have to text someone that you'll call them back if they call you first because they won't be able to hear you.

(Phosh 20200629 Release on PinePhone UBports CE)

Danct12, 2020-07-02:
I'll look for a fix after I receive my UBPorts CE, my dev phone modem isn't working.

philmmanjaro, 2020-07-10:
We fixed that on our end by patching [wys](https://gitlab.manjaro.org/manjaro-arm/packages/community/phosh/wys).

Danct12, 2020-07-11:
Fixed in Huong Tram Linux branch.

Danct12, 2020-07-11:
Fixed as in 2020/07/11 image.

closed 2020-07-11 by Danct12