dreemurrs-embedded/Pine64-Arch #6: Telegram bug

ScardracS, 2020-07-01:
Not sure about other OS, but there is a bug that prevent Telegram to pass first screen if sound is set as "Make a voice call" instead of "Play HiFi". I will make further searches in order to determine if it's related by alarm (as well as other OS, if problem appears) or if is a problem of Telegram.

P.S.

I really hope this is a bug and not a Telegram "feature".

Danct12, 2020-07-02:
Probably a bug with PulseAudio.

prurigro, 2020-07-10:
I'm not getting this issue myself-- both output and input are set to "Make a phone call" and telegram desktop is working correctly

Danct12, 2020-07-11:
Tested, I cannot reproduce the issue on my phone, I'll mark it as invalid now, please reopen if needed.

closed 2020-07-11 by Danct12

ScardracS, 2020-07-12:
You can sign it as fixed