dreemurrs-embedded/Pine64-Arch #8: Can't update with pacman -Syu.

vgnmnky, 2020-07-04:
Using 20200629.  Connected to internet via wifi.  

su, then Pacman -Syu, fails to resolve the urls, therefore fails to retrieve any of the db files, etc.

I know other people must have got this working, but I can't see a fix mentioned for this.  Probably straightforward, but might be helpful for people to find the answer on here.

Fails in software centre too.

Danct12, 2020-07-04:
This is already known, it's systemd-resolved.

You need to do
`sudo systemctl disable --now systemd-resolved`
Then enable it again and fingers crossed you'll have full WiFi access.
`sudo systemctl enable --now systemd-resolved`

I'll stop using systemd-resolved and moving to NetworkManager in the next image.

vgnmnky, 2020-07-04:
Fantastic, thanks for the quick response, that's fixed it!

Danct12, 2020-07-05:
Solved in the new image.

closed 2020-07-05 by Danct12