dreemurrs-embedded/Pine64-Arch #9: Some PinePhone batches has unstable DRAM controller

Danct12, 2020-07-05:
I have the 1.1 and have terrible crashes with all updated speeds. There was testing going on in the pine64 forums to find out about usable memory speeds to control the crashes. 

Though, the crashes with your arch are not as bad as they were on pmOS, I never got to stable speeds with any of the original test speeds all the way down to 512. However, for the few of us that never got stable, a new version was released with 492MHz speed and it seemed to work for us. 

Some of the unlucky braveheart users have hardware that cant run any level of up speed for the ram and it causes random crashes that lock up the whole system. Sometimes it does a soft crash and restart. But sometimes, it just freezes completely until you hard power off the device as well.

Here's a link to the forum post I made on my freezes in pmOS: https://forum.pine64.org/showthread.php?tid=9891

And this is the post of the guy testing memory speeds, where in some of the later pages he adds a 492 speed for people like me and a few others who cant run anything above base due to losing the silicon lottery: https://forum.pine64.org/showthread.php?tid=9832

_Originally posted by @crwagn in https://github.com/dreemurrs-embedded/Pine64-Arch/issues/3#issuecomment-653841207_

crwagn, 2020-07-05:
Okay, just tried the new release, can definitely confirm the same issue is still occurring. I see you added a new crash screen that popped up saying that something went wrong and I needed to log out, except it popped up while I was trying to put in the pass code to first login after boot up. After a minute or 2 of a black screen it did do a restart, and hopefully this time I can log in (it's inconsistent when it decides to crash with memory).

Danct12, 2020-07-05:
> I see you added a new crash screen that popped up saying that something went wrong and I needed to log out

We did not add anything, that's more like gnome-session crashed.

Anyway, I think you can try to install another U-Boot that has a lower speed, that should work for you.

crwagn, 2020-07-05:
I'll try that out, but I would also suggest having one version of your release that doesn't have the upped memory speed since I don't think many builds of uboot have the boot speed and power advancements you have added in your build. In fact, last I saw, I think the current memory testing one for 492MHz speed doesn't have working deep sleep for some reason.

Also, just by the survey, it looks like 50% of the testers couldn't run this high a speed of 624MHz stably anyway (though it is a skewed survey). So I just figured I'd throw that out. I know the performance gains are great, so I wouldn't ask you to get rid of this version, just a slower version for those of us who can't win in the silicon lottery so you can get more eyes and hands on your release for testing (if that's what you want).

Thanks again for the help and the awesome Arch releases!

Danct12, 2020-07-06:
I'll release a separate U-Boot image and have them to manually flash it after the rootfs flash, as producing new images can be a pain in upload times.

crwagn, 2020-07-06:
That's completely understandable, and thank you for your help. I definitely tried the traditional way of doing the uboot overwrite and it didn't work correctly, so I was starting to look into having to do a build from scratch. I'm happy to hear that others will not have to go through that to try this awesome distro!

Thanks again!

Danct12, 2020-09-03:
The latest U-Boot should provide 624, 552, 492 for both PinePhone and PineTab.

closed 2020-09-03 by Danct12

crwagn, 2020-09-03:
Thank you very much!

You will likely only need 552 and 624 shortly, the power savings seem to give more stability from my testing, I got all the way up to 576 before crashing now. But I have still seen a few on the forum complaining of instability on mobian, and I believe they use 552, so you are probably better off holding 492 for a few updates before removing it.

Thank you again for your hard work!