dreemurrs-embedded/Pine64-Arch #11: crust doesn't work correctly on PinePhone v1.1 (Braveheart)

Danct12, 2020-07-08:
One of the braveheart user from the Telegram chat mentioned that once the device goes sleep, it never wakes up.

Currently this issue has already been reported on:
- Mobian: https://gitlab.com/mobian1/devices/pinephone-linux/-/issues/8
- Upstream Pine64 Kernel: https://gitlab.com/pine64-org/linux/-/issues/24
- postmarketOS: https://gitlab.com/postmarketOS/pmaports/-/merge_requests/1163#note_369492589

So, we need to look into it.

Danct12, 2020-07-11:
Assigned to @asriel-danctnix as he has a BH v1.1.

Danct12, 2020-07-12:
Reported to happen on PinePhone v1.0, and it only happens without the charging cable plugged in, otherwise it works just fine.

Looking into this now.

justinesmithies, 2020-07-25:
My BH suffers from this too but doctoring the pinephone-suspend-hook.sh fixes it. The BH then wakes everytime after a deep sleep with no charger cable plugged in.

`#!/bin/sh

# Copyright 2020 - Dreemurrs Embedded Labs / DanctNIX Community
#
# This suspend hook is intended as a workaround for Bluetooth suspend oops:
# https://gitlab.com/pine64-org/linux/-/issues/24

prepare_suspend() {
	echo "Preparing to suspend..."
	echo "1c19000.usb" > /sys/bus/platform/drivers/musb-sunxi/unbind
	echo serial0-0 > /sys/bus/serial/drivers/hci_uart_h5/unbind
}

resume_all() {
	echo "Resuming the device..."
	sleep 1
	echo "1c19000.usb" > /sys/bus/platform/drivers/musb-sunxi/bind
	echo serial0-0 > /sys/bus/serial/drivers/hci_uart_h5/bind
	hwclock -s
}

case $1 in
	pre) prepare_suspend ;;
	post) resume_all ;;
esac`

ConnorGutman, 2020-07-25:
Copying the modifications made by @justinesmithies to /usr/lib/systemd/system-sleep/pinephone-suspend-hook.sh and giving the file a chmod of 755 fixed suspend for my BH as well! TBH these changes are a bit out of my realm of knowledge, any idea if this fix will alter the performance of crust?

retro486, 2020-08-11:
Looks like this fix made it into the latest image as a variable in pinephone-suspend-hook.sh. I can't find the commit to tag it here but I followed the instructions in the script on my BH and it works great to fix the no wake issue.

Danct12, 2020-11-13:
This might have been fixed with the latest 5.9 kernel (2020/11/12).

Feel free to reopen the issue if this is still happening with the latest build.

closed 2020-11-13 by Danct12