dreemurrs-embedded/Pine64-Arch #12: alsa-ucm-pinephone 0.1-1 breaks audio

prurigro, 2020-07-09:
On 0.1-0 you get the correct set of output and input devices and configurations, but on 0.1-1 you get "multichannel output - built-in audio" and "headphones", and neither of these output audio to the speakers or headphones. Similarly, input is limited in options and also doesn't work.

The regression seems to match the issue described here https://gitlab.com/postmarketOS/pmaports/-/issues/534 except in this case it seems to be because the name of the folder changed from sun50i-a64-audi to PinePhone.

Danct12, 2020-07-09:
It's changed for the new kernel (5.7), and also it'll make it easier to unify a rootfs.

If you're still on 5.6, you can cherry-pick these patches from Mobian which should fix it:
https://gitlab.com/mobian1/devices/pinephone-linux/-/commits/pinephone-5.6

prurigro, 2020-07-09:
Ahh I don't know why I didn't stop to think whether it was an issue with the kernel version-- I guess they usually try to avoid breaking userspace. Thanks for the explanation and course of action :)

closed 2020-07-09 by prurigro