dreemurrs-embedded/Pine64-Arch #14: 'gnome-usage' segfaults as normal user

philmmanjaro, 2020-07-10:
For some unknown reasons `gnome-usage` segfaults with normal user. You can start it however with **root**. Works however, when using **Mobian**. Also on our end with Manjaro we can't get it to run as normal user. For testing we recommend the mobile version: [gnome-usage-mobile](https://gitlab.manjaro.org/manjaro-arm/packages/community/phosh/gnome-usage-mobile)

philmmanjaro, 2020-07-19:
Does this make any sense?

```
[manjaro@manjaro-arm ~]$ G_MESSAGES_DEBUG=all gnome-usage                       
(gnome-usage:4690): GLib-GIO-DEBUG: 21:38:37.194: _g_io_module_get_default: Foun
d default implementation dconf (DConfSettingsBackend) for ?gsettings-backend?   
(gnome-usage:4690): dconf-DEBUG: 21:38:37.197: watch_fast: "/org/gnome/desktop/i
nterface/" (establishing: 0, active: 0)                                         
(gnome-usage:4690): dconf-DEBUG: 21:38:37.198: watch_fast: "/org/gnome/settings-
daemon/peripherals/mouse/" (establishing: 0, active: 0)                         
(gnome-usage:4690): dconf-DEBUG: 21:38:37.198: watch_fast: "/org/gnome/desktop/s
ound/" (establishing: 0, active: 0)                                             
(gnome-usage:4690): dconf-DEBUG: 21:38:37.199: watch_fast: "/org/gnome/desktop/p
rivacy/" (establishing: 0, active: 0)                                           
(gnome-usage:4690): dconf-DEBUG: 21:38:37.199: watch_fast: "/org/gnome/desktop/w
m/preferences/" (establishing: 0, active: 0)                                    
(gnome-usage:4690): dconf-DEBUG: 21:38:37.199: watch_fast: "/org/gnome/settings-
daemon/plugins/xsettings/" (establishing: 0, active: 0)                         
(gnome-usage:4690): dconf-DEBUG: 21:38:37.199: watch_fast: "/org/gnome/desktop/a
11y/" (establishing: 0, active: 0)                                              
(gnome-usage:4690): dconf-DEBUG: 21:38:37.206: watch_established: "/org/gnome/de
sktop/interface/" (establishing: 1)                                             
(gnome-usage:4690): dconf-DEBUG: 21:38:37.207: watch_established: "/org/gnome/se
ttings-daemon/peripherals/mouse/" (establishing: 1)                             
(gnome-usage:4690): dconf-DEBUG: 21:38:37.207: watch_established: "/org/gnome/de
sktop/sound/" (establishing: 1)                                                 
(gnome-usage:4690): dconf-DEBUG: 21:38:37.207: watch_established: "/org/gnome/de
sktop/privacy/" (establishing: 1)                                               
(gnome-usage:4690): dconf-DEBUG: 21:38:37.208: watch_established: "/org/gnome/de
sktop/wm/preferences/" (establishing: 1)                                        
(gnome-usage:4690): dconf-DEBUG: 21:38:37.208: watch_established: "/org/gnome/se
ttings-daemon/plugins/xsettings/" (establishing: 1)                             
(gnome-usage:4690): dconf-DEBUG: 21:38:37.209: watch_established: "/org/gnome/de
sktop/a11y/" (establishing: 1)                                                  
(gnome-usage:4690): Gdk-DEBUG: 21:38:37.327: Compositor prefers decoration mode 
'client'                                                                        
(gnome-usage:4690): GLib-GIO-DEBUG: 21:38:37.341: _g_io_module_get_default: Foun
d default implementation local (GLocalVfs) for ?gio-vfs?                        
(gnome-usage:4690): GLib-DEBUG: 21:38:37.501: unsetenv() is not thread-safe and 
should not be used after threads are created                                    
(gnome-usage:4690): Gtk-DEBUG: 21:38:37.502: Connecting to session manager      
(gnome-usage:4690): dconf-DEBUG: 21:38:37.715: watch_fast: "/org/gnome/Usage/" (
establishing: 0, active: 0)                                                     
(gnome-usage:4690): dconf-DEBUG: 21:38:37.717: watch_established: "/org/gnome/Us
age/" (establishing: 1)                                                         
                                                                                
** (gnome-usage:4690): WARNING **: 21:38:37.793: game-mode.vala:95: GameMode D-B
us error: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name com.fe
ralinteractive.GameMode was not provided by any .service files                  
** (gnome-usage:4690): DEBUG: 21:38:37.797: system-monitor.vala:146: new_len: 18
8, old_len: 0                                                                   
                                                                                
** (gnome-usage:4690): DEBUG: 21:38:37.798: system-monitor.vala:178: process add
ed: 1                                                                           
                                                                                
** (gnome-usage:4690): DEBUG: 21:38:37.798: system-monitor.vala:178: process add
ed: 2                                                                           
                                                                                
** (gnome-usage:4690): DEBUG: 21:38:37.799: system-monitor.vala:178: process add
ed: 3                                                                           
                                                                                
** (gnome-usage:4690): DEBUG: 21:38:37.799: system-monitor.vala:178: process add
ed: 4                                                                           
                                                                                
** (gnome-usage:4690): DEBUG: 21:38:37.800: system-monitor.vala:178: process add
ed: 6                                                                           
                                                                                
** (gnome-usage:4690): DEBUG: 21:38:37.801: system-monitor.vala:178: process add
ed: 8                                                                           
                                                                                
[... 597 lines cut ...]
[... cut ...]
size = 250'                                                                     
Tracker-INFO: 21:38:40.023:   Setting cache size to 250                         
(gnome-usage:4690): dconf-DEBUG: 21:38:40.039: watch_fast: "/org/freedesktop/tra
cker/fts/" (establishing: 0, active: 0)                                         
(gnome-usage:4690): dconf-DEBUG: 21:38:40.041: watch_established: "/org/freedesk
top/tracker/fts/" (establishing: 1)                                             
Segmentation fault (core dumped)    
```

nickgirga, 2020-07-23:
Launching under fakeroot at least allows you to manage your processes (including killing them) and view your resource usage. This solution only has one issue that I can narrow down to the usage of fakeroot: the lack of user labels (and the fact that it keeps complaining about it in the terminal). But it works for what I need it to right now—quickly tapping an app and killing some processes—so I've just kept using it.

The warnings (tested on Manjaro):
`app-item.vala:158:Unable to obtain user account: Error sending credentials: Error sending message: Operation not permitted`

There are still some other things not functioning properly, but they weren't functioning properly for me running the app authentically as root anyways (like the Storage tab seemingly taking forever to load and the Temperature tab needing gthd (which has refused to launch due to failure to acquire 'org.sigxcpu.Thermal' service name)).

Danct12, 2020-08-10:
Backtrace:

```
Thread 1 "gnome-usage" received signal SIGSEGV, Segmentation fault.
[Switching to LWP 5312]
0x0000007fb7c45ad8 in g_path_get_basename (
    file_name=0xd0b8d09cd03bb0d0 <error: Cannot access memory at address 0xd0b8d09cd03bb0d0>)
    at ../glib/gfileutils.c:2365
2365    ../glib/gfileutils.c: No such file or directory.
(gdb) bt
#0  0x0000007fb7c45ad8 in g_path_get_basename (
    file_name=0xd0b8d09cd03bb0d0 <error: Cannot access memory at address 0xd0b8d09cd03bb0d0>)
    at ../glib/gfileutils.c:2365
#1  0x0000005555578878 in usage_process_get_full_process_cmd (pid=pid@entry=3) at ../src/process.vala:116
#2  0x0000005555578afc in usage_process_construct (object_type=<optimized out>, pid=pid@entry=3)
    at ../src/process.vala:49
#3  0x0000005555579350 in usage_process_new (pid=pid@entry=3) at ../src/process.vala:46
#4  0x0000005555583b38 in usage_system_monitor_update_data (self=self@entry=0x7fb304f3e0)
    at ../src/system-monitor.vala:161
#5  0x0000005555583db8 in usage_system_monitor_init (self=0x7fb304f3e0) at ../src/system-monitor.vala:97
#6  0x0000005555583f6c in usage_system_monitor_construct (object_type=<optimized out>) at ../src/system-monitor.vala:74
#7  0x0000005555584148 in usage_system_monitor_new () at ../src/system-monitor.vala:61
#8  0x0000005555584164 in usage_system_monitor_get_default () at ../src/system-monitor.vala:46
#9  0x0000005555576b34 in usage_process_list_box_construct (object_type=<optimized out>, 
    type=type@entry=USAGE_PROCESS_LIST_BOX_TYPE_PROCESSOR) at ../src/process-list-box.vala:56
#10 0x00000055555770bc in usage_process_list_box_new (type=type@entry=USAGE_PROCESS_LIST_BOX_TYPE_PROCESSOR)
    at ../src/process-list-box.vala:38
#11 0x000000555556d5cc in usage_processor_sub_view_construct (object_type=<optimized out>)
    at ../src/cpu-sub-view.vala:43
#12 0x000000555556d858 in usage_processor_sub_view_new () at ../src/cpu-sub-view.vala:28
#13 0x00000055555757d8 in usage_performance_view_construct (object_type=<optimized out>)
    at ../src/performance-view.vala:51
#14 0x0000005555575978 in usage_performance_view_new () at ../src/performance-view.vala:45
#15 0x0000005555586ab8 in usage_window_construct (object_type=<optimized out>, 
    application=application@entry=0x7fb3dae100) at ../src/window.vala:79
#16 0x0000005555586f1c in usage_window_new (application=application@entry=0x7fb3dae100) at ../src/window.vala:59
#17 0x000000555556a158 in usage_application_real_activate (base=0x7fb3dae100) at ../src/application.vala:52
#18 usage_application_real_activate (base=0x7fb3dae100) at application.c:250
#19 0x0000007fb7d3aed0 in g_type_class_meta_marshalv (closure=<optimized out>, return_value=<optimized out>, 
--Type <RET> for more, q to quit, c to continue without paging-- 
    instance=<optimized out>, args=..., marshal_data=<optimized out>, n_params=<optimized out>, 
    param_types=<optimized out>) at ../gobject/gclosure.c:1034
#20 0x0000007fb7d3ca6c in _g_closure_invoke_va (closure=0x7fb6ce83e0, return_value=0x0, instance=0x7fb3dae100, 
    args=..., n_params=0, param_types=0x0) at ../gobject/gclosure.c:873
#21 0x0000007fb7d50db8 in g_signal_emit_valist (instance=instance@entry=0x7fb3dae100, signal_id=<optimized out>, 
    detail=detail@entry=0, var_args=...) at ../gobject/gsignal.c:3407
#22 0x0000007fb7d516c8 in g_signal_emit (instance=instance@entry=0x7fb3dae100, signal_id=<optimized out>, 
    detail=detail@entry=0) at ../gobject/gsignal.c:3554
#23 0x0000007fb7e63e18 in g_application_activate (application=application@entry=0x7fb3dae100)
    at ../gio/gapplication.c:2303
#24 0x0000007fb7e640ec in g_application_real_local_command_line (application=0x7fb3dae100, arguments=0x7ffffffb48, 
    exit_status=0x7ffffffb44) at ../gio/gapplication.c:1139
#25 0x0000007fb7e64338 in g_application_run (application=0x7fb3dae100, argc=<optimized out>, argv=0x7ffffffbb8)
    at ../gio/gapplication.c:2528
#26 0x000000555556d934 in _vala_main (args=0x7ffffffbb8, args_length1=1) at ../src/gnome-usage.vala:29
#27 0x0000007fb7f70914 in ?? () from /lib/ld-musl-aarch64.so.1
#28 0x0000007fb7fffa10 in ?? () from /lib/ld-musl-aarch64.so.1
#29 0x0000007ffffffdee in ?? ()
Backtrace stopped: not enough registers or memory available to unwind further
```

Danct12, 2020-08-15:
Seems to have fixed on it's own with the latest image.

closed 2020-08-15 by Danct12