dreemurrs-embedded/Pine64-Arch #16: how to add other locale (language)?

kingemir, 2020-07-16:
The gnome settings has a topic "region & language" where one could choose his language. But there's not much to choose from, only "Unspecified [ANSI_X3.4-1968]"

Danct12, 2020-07-17:
You need to edit /etc/locale.gen and uncomment which language you need in UTF-8. Save, then run locale-gen as root, which after that you should be able to pick a new language.

ScardracS, 2020-07-17:
@Danct12 that's not works (tested with Italian). Language still english

sebastichoupinet, 2020-07-18:
@Danct12 @ScardracS 
A bit more infos, after uncommenting locale.gen:
parts of gnome are still in english : settings, clocks, terminal keyboard 
parts are in my language (french): global search, input source
Before uncommenting, even if I could select another input source, it never loaded.

kingemir, 2020-07-18:
I can confirm that after editing the /etc/locale.gen and running locale-gen, the added languages show up. But the same as @ScardracS and @sebastichoupinet described, after choosing one of them in the settings and do reboot (over terminal, since "Restart" buttom doesn't work), the interface stays in english.

ScardracS, 2020-07-19:
@sebastichoupinet where I can help on translating?

sebastichoupinet, 2020-07-19:
@ScardracS @Danct12  
Don't know :) But I can help too, in spanish and french if it's needed .

Danct12, 2020-08-02:
Locale issue fixed in phosh 0.4.2-3.

closed 2020-08-02 by Danct12