dreemurrs-embedded/Pine64-Arch #19: GNOME/Web (Epiphany) doesn't actually play youtube videos

godofgrunts, 2020-07-20:
The release for the July 15 release says:

* Fixed YouTube playback on Epiphany

However, I'm still getting "Your browser does not currently recognize any of the video formats available.

I even tried reinstalling `gstreamer`gst-plugins-good` and `gst-plugins-bad` but to no avail.

Danct12, 2020-07-22:
weird, seems to work for me just fine since that day i tested.

i'll check it out later when i got back home

Danct12, 2020-07-28:
Was unable to produce the issue with the latest image..

[image]

asriel-danctnix, 2020-07-29:
I was able to get it working by just installing gst-plugins-bad.

ScardracS, 2020-07-31:
Yeah, gst is a must in order to make it working. If you try a clean build without it doesn't work

Danct12, 2020-08-17:
Fixed by installing gst-plugins-bad, will be a part of next image.

Also, since Epiphany is deprecated, please check out mobile-config-firefox, you'll like it.

closed 2020-08-17 by Danct12