dreemurrs-embedded/Pine64-Arch #21: cheese replacement?

Tarcaxoxide, 2020-07-28:
so i noticed you removed cheese so i was curious why . i installed it and found out quickly that it could not detect a device.
so then i installed mpv and checked the /dev/video* files.
/dev/video0 doesn't seem to open but /dev/video1 appears to be the back camera.
mpv /dev/video1 works but its fairly lagy.

i don't know if this helps any?

Tarcaxoxide, 2020-07-28:
the video feed also appears to be rotate 90 degrees

Danct12, 2020-07-29:
gnome-camera is the current replacement.

Tarcaxoxide, 2020-07-29:
> gnome-camera is the current replacement.

that doesn't even seem to open

``
[alarm@are-you-winning-son ~]$ gnome-camera 
[E][000000179.072536][core.c:71 core_event_error()] core 0xaaaaf33f5100: proxy 0xffff8c002e00 id:7: seq:13 res:-25 (Inappropriate ioctl for device) msg:"enum params id:3 (Spa:Enum:ParamId:EnumFormat) failed"
[E][000000179.072656][gstpipewiredeviceprovider.c:326 on_core_error()] error id:7 seq:13 res:-25 (Inappropriate ioctl for device): enum params id:3 (Spa:Enum:ParamId:EnumFormat) failed
``

asriel-danctnix, 2020-07-29:
Camera is currently broken after the HDMI patch was merged, so this needs to look further.

Tarcaxoxide, 2020-07-29:
i'd take a picture but your running wayland .

i don't know of a way to x11forward that