dreemurrs-embedded/Pine64-Arch #22: pinephone 1.2 after pin entry correct ui is shown then screen fades to black and only mouse shows

shockron22, 2020-07-31:
When I boot up my pine phone for the first time and enter my pin. The app tray shows up correctly then the screen fades away to black. When I connect a mouse only the pointer shows up on the screen. I can ctrl-alt-f2 into another terminal session and it seems to work normally. Having a hard time tracking down what exactly is causing the black screen to popup over the normal ui. Sometimes if I lock and unlock the phone the black screen goes away and I can use the phone as normal. This happens every time a reboot or turn the phone on. Any pointers would be helpful. Thanks!

brianmcarey, 2020-08-14:
Same issue has been seen in Mobian - https://gitlab.com/mobian1/issues/-/issues/3

Danct12, 2020-09-19:
This might be a hardware flaw, but not so many people have the issue for now I'll be closing this.

closed 2020-09-19 by Danct12