dreemurrs-embedded/Pine64-Arch #25: Input sources not all working?

kingemir, 2020-08-04:
I tried the input sources _German_ and _French_, they seem to work. 
What didn't work were _German (Switzerland)_ and _French (Switzerland)_ (Both look just like generic English, but they should have a QWERTZ layout like the German layout, and some ways to type Umlaut and Accents should be offered).
Also input sources _Arabic_ and _Japanese_ just show the generic English layout, without any input methods applied nor any language specific onscreen keyboard layout.

ScardracS, 2020-08-04:
Not every locales are set up ATM. You will need for further update from phosh

Danct12, 2020-09-08:
Seems to be fixed, here is a list of all input sources from squeekboard: https://source.puri.sm/Librem5/squeekboard/-/tree/master/data/keyboards

closed 2020-09-08 by Danct12