dreemurrs-embedded/Pine64-Arch #26: HDMI output crashes braveheart with usb-c fix?

ConnorGutman, 2020-08-05:
Just fixed the USB-C hardware issue on my Pinephone braveheart edition and flashed the latest Arch build. USB devices such as keyboard+mouse as well as power passthrough are working with a dock I own ([from Amazon](https://www.amazon.com/dp/B07Q6YS7W2/ref=cm_sw_r_cp_apa_i_gdIkFbKM7F6HM)) but HDMI output seems to crash the device. Upon crashing the device shows the boot animation and restarts Phosh. I've attempted this with a few monitors as well as a TV. Not sure if it's the particular usb-C dock I'm using so I'm buying a few other ones to test with, but I'm curious if anyone else is having this issue or if perhaps I'm missing a step? I would appreciate any tips or suggestions from anyone else who has working HDMI output, thanks!

Danct12, 2020-08-05:
This seems to be a common issue with wlroots, I will have to recompile phoc with debug later

aeazocar, 2020-08-13:
I have found almost the same problems as shockron22:
What I am using:
- Pinephone Bravehart 1.1 (with raw USB-C hardware fix - removed the components only)
- NexDock2 USB-C dock
- Generic USB-C dock + Keyboard + HDMI display

With the NexDock2:
1. I have the "HDMI output crashes Phoc #26"
2. If I time it correctly I can get the pinephone to display the GUI.
Note: The NexDock2 has a very short shutdown sequence, if no Display signal is detected with in few seconds, it will power off completely (no keyboard and / or mousepad operation

With Generic USB-C dock + Keyboard + HDMI display: (From shockron22 post in "quotes")
1. "When I boot up my pine phone for the first time and enter my pin. The app tray shows up correctly then the screen fades away to black."
2. "When I connect a mouse only the pointer shows up on the screen."
3. "I can ctrl-alt-f2 into another terminal session and it seems to work normally."
4. I can not get to the phone GUI at all.

Regards

Danct12, 2021-03-31:
So far, I have not heard about this anymore and it's been a while. I'll be closing this issue.

Feel free to reopen if this is still a issue.

closed 2021-03-31 by Danct12