dreemurrs-embedded/Pine64-Arch #28: Terminal hangs when an invalid ctrl + <key> sequence is used on the on-screen keyboard.

rvndpr, 2020-08-09:
This is consistently reproducible. Key in something at the terminal prompt. Hit the ctrl key followed by a key that is not a valid sequence (eg. ctrl+s). The terminal hangs. I tried to see if it entered vi escape mode or something like that - but that was not the case. I had to close the terminal and open a new instance of the terminal.

I was running this arch distro from an SD card on my pinephone UBPorts Community Edition.

zermingore, 2020-08-17:
Sorry for asking but does it still hang after you send a Ctrl-q ?
(just to make sure the readline is not guilty)

Danct12, 2020-08-17:
I can reproduce this issue, this is a squeekboard bug I guess.