dreemurrs-embedded/Pine64-Arch #29: Cell connectivity completely stops after hours of deep sleep

nickgirga, 2020-08-10:
I can't figure out specifically what triggers it as it doesn't happen if I wake it only an hour or two after deep sleep. Once noticed, I set deep sleep to turn on when plugged in for testing. It consistently happens when I leave in on the charger when I go to bed now. When I wake up in the morning, the mobile icon in the status bar is missing and the settings page for mobile completely disappears. So it appears to only happen after LONG periods of time without activity. Temporarily fixed with a reboot.

Tested on UBports CE (v1.2) on the last released image with all updates applied as of Aug 9, 8:30PM PST.

I've attached 2 essentially identical screenshots that demonstrate the behavior after waking the device after hours of inactivity.

[image]
[image]

Danct12, 2020-08-13:
can you give me output of lsusb when this happened? you should see a usb device something like "Qualcomm ..."

nickgirga, 2020-08-13:
`Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub`
`Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub`
`Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

This is all I got. After restarting and regaining connectivity, `Bus 003 Device 002: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem` shows up with everything else remaining the same.

juanantoniomosquera, 2020-10-06:
same issue

Pacman99, 2020-10-14:
I have the same issue, I can't call or text after a deep sleep, the modem is just gone. It doesn't show up in lsusb or mmcli.

I found some lines in dmesg that seem relevent: 
```
[ 2851.738935] usb usb3-port1: disabled by hub (EMI?), re-enabling...
[ 2851.745298] usb 3-1: USB disconnect, device number 2
[ 2851.746485] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[ 2851.746715] option 3-1:1.0: device disconnected
[ 2851.757905] option1 ttyUSB2: usb_wwan_open: submit int urb failed: -19
[ 2851.764576] option1 ttyUSB2: usb_wwan_open: submit read urb 0 failed: -19
[ 2851.769326] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[ 2851.772773] option1 ttyUSB2: usb_wwan_open: submit read urb 1 failed: -19
[ 2851.776706] option 3-1:1.1: device disconnected
[ 2851.780613] option1 ttyUSB2: usb_wwan_open: submit read urb 2 failed: -19
[ 2851.788272] option1 ttyUSB2: usb_wwan_open: submit read urb 3 failed: -19
[ 2851.812332] option1 ttyUSB2: GSM modem (1-port) converter now disconnected from ttyUSB2
[ 2851.815899] option 3-1:1.2: device disconnected
[ 2851.842203] option1 ttyUSB3: GSM modem (1-port) converter now disconnected from ttyUSB3
[ 2851.842324] option 3-1:1.3: device disconnected
[ 2851.843260] qmi_wwan 3-1:1.4 wwan0: unregister 'qmi_wwan' usb-1c1b000.usb-1, WWAN/QMI device
```

Pacman99, 2020-10-26:
I found a simple workaround for this. Whenever this happens you can just restart the modem and it gets picked up again.
Run these as root:
`echo 0 > /sys/class/modem-power/modem-power/device/powered` 
wait a couple seconds then `echo 1 > /sys/class/modem-power/modem-power/device/powered`.

I got the commands from https://github.com/dreemurrs-embedded/Pine64-Arch/blob/master/PKGBUILDS/pine64/danctnix-eg25-misc/eg25_power.sh.

MichaelArthurLong, 2020-11-17:
@Pacman99 or better yet, just use the script itself.

Danct12, 2020-11-18:
Seems like the issue is gone with the new modem firmware.

For more information on how to upgrade the modem firmware, check out this [GitHub repository](https://github.com/Biktorgj/quectel_eg25_recovery).  **Do this at your own risk!**

closed 2020-11-18 by Danct12

Pacman99, 2020-11-18:
@Danct12 I saw that github repo and subsentients forum post on updating the modem firmware. Is it necessary to put the modem into qdl/edl mode? The github repo says to do it but subsentient doesn't mention it.

Danct12, 2020-11-18:
@Pacman99 Yes, you need to put the modem to EDL (either using test points or `adb reboot edl`)

Pacman99, 2020-11-18:
> @Pacman99 Yes, you need to put the modem to EDL (either using test points or `adb reboot edl`)

Okay thank you!