dreemurrs-embedded/Pine64-Arch #32: Incoming SMS no longer being received

nigelnorman, 2020-08-15:
My device has not been able to receive any incoming SMS messages. I have verified that outgoing messages are being reached by their intended recipients, as well as all incoming and outgoing calls seem to continue to go through.

When taking my sim out and putting it into another device, all incoming messages that were not received by will suddenly come through at once.

This issue first seems to have started after applying updates on Aug 10, 12:00PM EST.

Using UBports CE (v1.2) on image 20200802 with all updates applied as of Aug 15, 10:00AM EST.

ScardracS, 2020-08-17:
> My device has not been able to receive any incoming SMS messages. I have verified that outgoing messages are being reached by their intended recipients, as well as all incoming and outgoing calls [...]

You should flash latest image. Unofrtunately the OS is under active development and not every changes is easily fixable with pacman. I have tried with it and no issue on mine

Danct12, 2020-08-17:
I also could not receive SMS on my device, looking into it.

Danct12, 2020-08-26:
I cannot reproduce this with the new sim card. Closing, please reopen if it still happens with the latest image!

closed 2020-08-26 by Danct12

nigelnorman, 2020-08-26:
flashed image 20200822 then updated via `sudo pacman -Syu ` and rebooted today. Confirmed Incoming and Outgoing calls are still working (better than before even) and all Outgoing SMS being received by other parties. Incoming SMS still not being received at all.

Danct12, 2020-08-27:
Weird, incoming SMS work fine for me.. not really sure what's happening here :\

nigelnorman, 2020-08-27:
I did try it using another sim card to be sure, and had the same results. Network defaults to 2G/3G/4G (preferred) with mobile data enabled as well. Both sims were with the same carrier: T-Mobile.

nigelnorman, 2020-08-27:
Now I'm suspecting it might be a hardware issue. I went back and flashed image 20200802 again, and did not run any updates yet, and still not coming through.

Thanks for looking into it, sorry to waste any time.

Danct12, 2020-08-27:
Could also be T-Mobile doing something weird, not sure. Can you it on Mobian and see if it has the same issue?

nigelnorman, 2020-08-27:
Yeah, so I did end up trying it with both Postmarket and Mobian, same results: incoming and outgoing calls working, and outgoing SMS working; incoming SMS still not working. At this point it really must be either something up with T-Mobile, or something is up with the device modem - but I find it bizarre that if that were the case that reception would not work while transmission does.

nigelnorman, 2020-08-27:
So I exercised my Google skills, and came across this:
https://forum.pine64.org/showthread.php?tid=9687&pid=64706#pid64706

aparently there are issues with the modem cache being full. I was attempting to do some of the solutions provided there; hadn't actually completed any, but all of a sudden SMS started coming through on Mobian.

So once again, I apologize for the false alarm by opening the issue. just leaving this here for info.

Thank you again for looking into it and support!

Danct12, 2020-08-28:
Good to know! I may try to develop a solution for this.