dreemurrs-embedded/Pine64-Arch #34: Lomiri or Glacier UX?

Qiangong2, 2020-08-17:
Is it possible to install Lomiri or Glacier UX on Arch for the pinephone?

I love all the features that are being implemented, but I really dislike Phosh

Danct12, 2020-08-20:
Plasma Mobile. I don't know, that'll be discussed very later!

charlesDelaware, 2021-01-06:
Simple X Mobile (sxmo) looks like it could be a a decent lighter weight option for the Pine Phone but, I haven't tested it out yet to see if their navigation set up is any good:
https://sr.ht/~mil/Sxmo/

Lin-Admin, 2021-01-16:
IMHO it is **much** too early to discuss this now.

Shatur95, 2021-02-04:
@Danct12, I noticed that plasma packages are already available in repo. Is it possible to install it manually on barebone image? Or are images with pre-installed plasma coming soon?

Danct12, 2021-02-05:
The latter.

alien2003, 2021-05-27:
Personally I would be really happy with Hildon/Matchbox UI ~~/ UX~~ from Maemo

Danct12, 2021-06-11:
I actually really like Maemo, but I'll probably use i3wm instead.

SAPllak, 2021-07-18:
> I actually really like Maemo, but I'll probably use i3wm instead when the keyboard case releases.

i really wish that arch had a lomiri desktop aswell

Silent-Hunter, 2021-08-25:
I'm really hoping SXMO gets added soon! It's my favorite.