dreemurrs-embedded/Pine64-Arch #41: locale-gen occurs everytime glibc updates

Danct12, 2020-09-03:
This takes a very VERY long time. So fix it.

ScardracS, 2020-09-03:
Personally I modify /etc/locale.gen in order to maintain only languages I really need (so en.US and it.IT).
Everyone should use that or add a script that permit to select only languages we are interested on (like localepurge)

Danct12, 2020-09-11:
Hmm, maybe the idea here is to generate glibc locale during buildtime and then ship it to users?

ScardracS, 2020-09-11:
@Danct12 probably better, but if an update occurs the problem still be here

Danct12, 2020-09-11:
We can disable that ability, there's no point to run locale-gen again once the locales are all installed.

Danct12, 2021-03-31:
So that is now fixed, I'll push out newer images that get around the problem.

But for older images, you need to install `glibc-locales` from the repository and clean `/etc/locale.gen`.

closed 2021-03-31 by Danct12