dreemurrs-embedded/Pine64-Arch #42: How to update uboot on pinephone

mogwa1, 2020-09-03:
I've updated uboot-pinephone through pacman.  However, the instructions to install the bootloader are not correct anymore.
It seems that the uboot-pinephone package now contains three versions of the u-boot-sunxi-with-spl-pinephone-xxx.bin file, with xxx being 492, 552 or 624.
Shouldn't the package contain just one version?

0p3t, 2020-09-03:
sudo dd if=/boot/u-boot-sunxi-with-spl-pinephone-624.bin of=/dev/mmcblk2 bs=8k seek=1 worked just fine for me. Guess the instructions on the updater need, well, updating.

mogwa1, 2020-09-03:
Ok, thanks.  I've just flashed this one, and it's fine now.
The main problem was that I didn't know which of the three should be installed, or what the difference is between the three versions.

Danct12, 2020-09-03:
These multiple versions are for devices that works better with lower DRAM clock. Usually if you have no issue then you'll be using 624. (see #9)

I'll add this to the wiki soon.

closed 2020-09-03 by Danct12