dreemurrs-embedded/Pine64-Arch #45: Default terminal

doublez13, 2020-09-05:
I'm a brand new user of Pine64-Arch, so forgive me if this is known or incorrect.

It looks like the default terminal value is set to gnome-terminal, which is not installed by default.
```
[root@p64 ~]# gsettings get org.gnome.desktop.default-applications.terminal exec
'gnome-terminal'
```
Installing a program that launches through the terminal (htop for example), creates an icon on the app screen, but won't lunch due to gnome-terminal not being installed. Installing gnome-terminal fixes the issue, but I would assume changing the exec value to the 'kgx' terminal would be the cleaner solution.

doublez13, 2020-09-05:
After exploring this further, it looks like Gnome does not respect that dconf key. It simply works it's way down a hard coded list of terminal emulators. Lame. 

https://gitlab.gnome.org/GNOME/glib/blob/master/gio/gdesktopappinfo.c#L2515

Danct12, 2021-06-24:
Superseded by #182

closed 2021-06-24 by Danct12