dreemurrs-embedded/Pine64-Arch #48: WAYLAND_DISPLAY env var not set

grelltrier, 2020-09-10:
I noticed that the environment variable WAYLAND_DISPLAY is not set when I am using phosh/phoc on Arch Linux on my Pinephone. On my desktop (also Arch) it is set to 'wayland-0' which I guess is the default. If an application tries using the environment variable to connect to a wayland server, it will fail. I noticed this because I am programming an application and it failed to connect. Can you possibly set this environment variable? I first reported this upstream to phoc but apparently it is supposed to get set.
Here is the issue I opened for phoc: [https://source.puri.sm/Librem5/phoc/-/issues/150](https://source.puri.sm/Librem5/phoc/-/issues/150)

lisael, 2020-09-18:
It is set for me, at least in a shell in the phone's terminal. Are you sure you're not trying to run your app from a ssh session? In this case, it's not set, and it's the intended behavior. You can just `export WAYLAND_DISPLAY=wayland-0` in your ssh shell session.

Danct12, 2020-09-26:
If you're using SSH, this is intended behavior. It is set in the phone's terminal for me.

closed 2020-09-26 by Danct12