dreemurrs-embedded/Pine64-Arch #50: No tap-to-click on touchpad

rakor, 2020-09-12:
There is no way to set a tap-to-switch for the kwyboard of the pinetab. Might be the mouse-settings on the gnome-settings app needs to be activated to set this feature on.

bolbishvili, 2020-09-12:
is it supported in this touchpad? I can click the touchpad  (push down on it) and it works, but not a light tap.

rakor, 2020-09-12:
> is it supported in this touchpad? I can click the touchpad (push down on it) and it works, but not a light tap.

That's the question we have to answer I think. Physically clicking is not that ergonomic to me.

bolbishvili, 2020-09-12:
I got it to work by doing the following:
`xinput list` 

to find the device, for me it was device 8, so next I

`xinput list-props 8`

 and saw that "libinput Tapping Enabled (281)" was set to 0, so to enable it I had to run

`xinput set-prop 281 1`

and now I can tap to click.  Of course, we don't want to do this manually every time, but we know that it's working.

rakor, 2020-09-13:
Here not working. I see no "libinput Tapping Enabled", for no device?!

wd5gnr, 2020-09-17:
This is another thing that works in Plasma (if you pick it in settings), so probably more of a Phosh issue. But yes, the hardware supports it.

Mashintime, 2020-09-24:
Tap to click does work with the touchpad.  I cheated and rebuilt libinput so it's doesn't default to disabling tap to click.  I think for this target audience making tap enabled by default wouldn't be an issue if deployed in the image.  Tap to click and two tap right click works.

I followed these build instructions:  https://wayland.freedesktop.org/libinput/doc/latest/building.html

Then edit tp_tap_default in src/evdev-mt-touchpad-tap.c to return LIBINPUT_CONFIG_TAP_ENABLED instead of LIBINPUT_CONFIG_TAP_DISABLED;`

There might be a better way (new to arch), but worked for me.

Danct12, 2022-02-14:
I'll close this as tap-to-click is disabled by default (same on my Pinebook Pro as well until I enable it in the settings).

closed 2022-02-14 by Danct12