dreemurrs-embedded/Pine64-Arch #55: No cell data available on Postmarket OS edition Pinephone (It is running arch not Postmarket OS) from Cricket Wireless

jprhyne, 2020-09-13:
When I turn my phone's mobile data on all cell signal is lost and can no longer receive nor send SMS messages. However, when I turn it off, signal returns shortly and SMS messages go through. My data works from android, so is not a cell plan issue. I have configured the APN to be name: "Cricket Internet" with APN: "ndo" as defined on their website.

Note: Connected over 4G with little 3G availability in my region. Let me know if there is anymore information needed.

I am not sure if this is a software or hardware issue since I cannot test with a linux os on another device.

mogwa1, 2020-09-14:
I have the exact same issue with a braveheart pinephone.  It had no problems connecting to 4G or 3G up to (I think) about two weeks ago.  So I have the impression that it could be related to an update that happened around that time.

When the phone boots, it connects to 4G almost instantaneously, but loses the connection after 10-20 seconds and never connects again until the next reboot.  It not only impacts data, it's also not possible to make phone calls or send/receive SMS.
I've also tried multiple APN settings etc.

Here's the ModemManager related output of journalctl.  Not how it initially connects and then drops the connection after a few minutes, after which it keeps repeating the same messages.
```
sep 14 09:51:18 pinephone systemd[1]: Starting Modem Manager...
sep 14 09:51:18 pinephone ModemManager[3603]: <info>  ModemManager (version 1.14.2) starting in system bus...
sep 14 09:51:18 pinephone systemd[1]: Started Modem Manager.
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0] Opening device with flags 'version-info, proxy'...
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0] created endpoint
sep 14 09:51:20 pinephone ModemManager[3603]: cannot connect to proxy: Could not connect: Connection refused
sep 14 09:51:20 pinephone ModemManager[3603]: spawning new qmi-proxy (try 1)...
sep 14 09:51:20 pinephone ModemManager[3603]: <warn>  [ttyUSB0/probe] failed to parse QCDM version info command result: -7
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0] Checking version info (20 retries)...
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0] QMI Device supports 30 services:
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    ctl (1.5)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    wds (1.67)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    dms (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    nas (1.25)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    qos (1.12)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    wms (1.10)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    auth (1.3)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    at (1.2)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    voice (2.1)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    cat2 (2.24)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    uim (1.46)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    pbm (1.4)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    test (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    loc (2.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    sar (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    ims (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    csd (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    ts (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    tmd (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    wda (1.16)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    csvt (1.1)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    imsp (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    imsa (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    coex (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    pdc (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    rfrpe (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    dsd (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    unknown [0x2d] (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    unknown [0x30] (1.0)
sep 14 09:51:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    unknown [0x36] (1.0)
sep 14 09:51:21 pinephone ModemManager[3603]: <info>  [device /sys/devices/platform/soc/1c1b000.usb/usb3/3-1] creating modem with plugin 'quectel' and '6' ports
sep 14 09:51:21 pinephone ModemManager[3603]: <info>  [base-manager] modem for device '/sys/devices/platform/soc/1c1b000.usb/usb3/3-1' successfully created
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0] Opening device with flags 'version-info, proxy'...
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0] created endpoint
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0] Checking version info (20 retries)...
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0] QMI Device supports 30 services:
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    ctl (1.5)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    wds (1.67)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    dms (1.0)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    nas (1.25)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    qos (1.12)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    wms (1.10)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    auth (1.3)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    at (1.2)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    voice (2.1)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    cat2 (2.24)
sep 14 09:51:21 pinephone ModemManager[3603]: [/dev/cdc-wdm0]    uim (1.46)
[... cut ...]
sep 14 09:55:36 pinephone ModemManager[3603]: <info>  [modem0] 3GPP Registration state changed (idle -> unknown)
sep 14 09:55:36 pinephone ModemManager[3603]: <warn>  [modem0] couldn't load operator code: Current operator MCC/MNC is still unknown
sep 14 09:55:36 pinephone ModemManager[3603]: <warn>  [modem0] couldn't load operator name: Current operator description is still unknown
sep 14 09:56:20 pinephone ModemManager[3603]: <info>  [sleep-monitor] system is about to suspend
sep 14 09:56:20 pinephone ModemManager[3603]: [/dev/cdc-wdm0] Releasing 'wds' client with flags 'release-cid'...
```

mogwa1, 2020-09-15:
In the meantime I found the cause of my particular problem.  It seems to be related to the nmconnection file which is generated by either gnome-control-center or NetworkManager.
I am roaming abroad with the SIM.  It seems that the gome-control-center is setting "network-id" explicitly in the nmconnection file, seemingly regardless of the roaming setting in the GUI.  If I manually remove this line from the nmconnection file and reload the config, then everything magically works again.
The only trouble now is that the control center is constantly overwriting this file...  Anyway, it seems to be an upstream problem.

I'm not sure that this solves @jprhyne 's problem, though.

Danct12, 2020-11-13:
I'll be closing the issue, feel free to reopen if it's still happening.

closed 2020-11-13 by Danct12