dreemurrs-embedded/Pine64-Arch #56: Proposal: Lower the CPU thermal throttling trip point

nadiadavis, 2020-09-13:
Some PinePhone users have reported under various circumstances that they have accidentally caused damage to their devices by overheating. While some of these reference issues that happened months ago and may no longer be relevant with software updates...

The default trip point is set to 75 degrees Celsius which is quite hot, especially when considering the display back panel lives right next to the CPU.

This can be adjusted via sysfs: `echo 58000 | sudo tee /sys/class/thermal/thermal_zone0/trip_point_0_temp` for 58 degrees, for example. I propose that we determine a reasonable default trip point.

It may be worth trying to collect data points from users about what temperature their phone reaches when it begins "feeling too hot" to the touch, etc.  There are also other configurable thermal zones with their own trip points to consider later.

nadiadavis, 2020-09-23:
Funeral on Discord reports that the operating temperature will not cause damage to the phone, however I believe there are still other concerns to consider:

- Phone feels hot to the touch
- The battery gets hot
- Temperatures over 45C can cause skin burns

fuchsmich, 2020-09-24:
There is also a thread about similar concerns on the [pine64 forum](https://forum.pine64.org/showthread.php?tid=11525) and two [blog](https://xnux.eu/log/#017) [posts](https://xnux.eu/log/#018) by megi.

Izaic, 2021-03-11:
With postmarketOS CE and newer pinephones you don't have to worry about thermals anymore. The added thermal pads to the mainboard and back cover mean that it only ever gets warm anymore.

Danct12, 2024-03-09:
It's been four years and it looks like my PinePhone is still working fine (most development is done on this device). Both the software and the battery itself has failsafe built in.

I'll be moving this to it's own discussion thread.