dreemurrs-embedded/Pine64-Arch #57: Make landscape the default orientation on PineTab

aqtrans, 2020-09-14:
Fantastic work, loving Arch on my PineTab! 

I cobbled together the following two patches to make the landscape orientation more of the default.

Now I'm not sure of how to integrate to allow PinePhone functionality as well, so I felt an Issue would be better than a PR. 

This replaces a part of [MR#554](https://source.puri.sm/Librem5/phosh/-/merge_requests/554), forcing a 90 degree orientation on startup: 
```
--- a/src/shell.c       2020-09-13 20:54:58.326528406 -0500
+++ b/src/shell.c       2020-09-13 20:54:31.266877995 -0500
@@ -462,9 +462,8 @@

   phosh_session_register (PHOSH_APP_ID);

-  /* If we start rotated, fix this up */
-  if (phosh_shell_get_rotation (self))
-    phosh_shell_rotate_display (self, 0);
+  // Try to default to landscape
+  phosh_shell_rotate_display (self, 90);

   priv->startup_finished = TRUE;
```

This undoes [MR#411](https://source.puri.sm/Librem5/phosh/-/merge_requests/411)
```
--- a/src/lockscreen-manager.c  2020-09-13 14:09:19.678181566 -0500
+++ b/src/lockscreen-manager.c  2020-09-13 14:11:50.067686831 -0500
@@ -72,9 +72,6 @@
   PhoshShell *shell = phosh_shell_get_default ();
   PhoshMonitorManager *monitor_manager = phosh_shell_get_monitor_manager (shell);

-  phosh_shell_rotate_display (shell, priv->rotation);
-  priv->rotation = 0;
-
   g_return_if_fail (PHOSH_IS_LOCKSCREEN (lockscreen));
   g_return_if_fail (lockscreen == PHOSH_LOCKSCREEN (priv->lockscreen));

@@ -161,10 +158,6 @@
   primary_monitor = phosh_shell_get_primary_monitor (shell);
   g_return_if_fail (primary_monitor);

-  /* Undo any rotation so the keypad becomes usable */
-  priv->rotation = phosh_shell_get_rotation (shell);
-  phosh_shell_rotate_display (shell, 0);
-
   /* Listen for monitor changes */
   g_signal_connect_object (monitor_manager, "monitor-added",
                            G_CALLBACK (on_monitor_added),
```

Danct12, 2020-09-15:
I don't think this can be easily fixed (the PineTab display is portrait). Again, it's just the lock screen so it shouldn't really matter much.

Once the device is unlocked, if you have auto rotation turned on it'll rotate the display to landscape automatically.

aqtrans, 2020-09-15:
Ahh yeah, on second thought you're right. :+1:  Just annoyed me enough to dig into :laughing:

closed 2020-09-15 by aqtrans