dreemurrs-embedded/Pine64-Arch #58: [Bug] Can't change Position for Set Desktop Background

sanecito, 2020-09-14:
HW: pmOS CE (3GB RAM)
SW: Arch Linux ARM (Phosh) - 2020/09/13 + post flash software store updates

Problem: When select holding an image on Firefox and selecting 'Set Desktop Background' you can't select any positions from the dropdown which means you can only use the default, 'Center'

Expectation: You can select any of the other Position options like 'Fit' with your finger

Replication:

1. Open Firefox and find an image on the web
2. Select hold image and select 'Set Desktop Background'
3. Select the Position dropdown
4. Try to select another option in the dropdown

ahoneybun, 2020-11-08:
I can't seem to recreate this on the latest image 10/24/20 as you can see in the attached photo. Have you tried the latest image/updates?

[image]

Danct12, 2020-11-29:
While this is infact a bug (workaround: switch to terminal kblayout and use arrow keys), Phosh currently does not support any other position other than fill.

closed 2020-11-29 by Danct12