dreemurrs-embedded/Pine64-Arch #59: Constant crashes and kernel panics on 3GB RAM version

mogwa1, 2020-09-17:
First of all, thank you for the effort you have spent to get a nice and polished experience on the pinephone.  I've been happily running your archlinuxarm version on a braveheart for a few months now.

I've recently received the 3GB version.  After re-flashing with your latest image, the device boots, but hangs within about 30 seconds.  Usually around the time it gets to PIN entry.  It's a bit better if I use the lower DRAM speed u-boot versions, but it still crashes, just a bit later.  And when I add an SD card and a SIM card, it's crashing again after about 30 seconds.
Actually, it seems there are several types of crashes happening:
- A kernel error.  In that case the display freezes, but the device keeps on running.  I can still log in with ssh and shut the device down properly.  I will add such a kernel log entry later.  Usually, when waiting long enough this will also lead to the second kind of crash.
- A hard crash.  The red LED is flashing, which I think indicates a kernel panic, right?

I've been looking around the forums and it looks like it could be related to missing u-boot support for the RAM: https://forum.pine64.org/showthread.php?tid=11236
I just wanted to ask whether the current u-boot version in the repository already has this patch in order to exclude this as the main reason.
If it already has this, then do you know what else could be the reason why I get such crashes?

Danct12, 2020-09-17:
We have provided 552 and 492 MHz as alternative DRAM speeds, please check the [release page](https://github.com/dreemurrs-embedded/Pine64-Arch/releases) and flash U-Boot to the target boot device.

For SD card, flash U-Boot using:
`dd if=u-boot-sunxi-with-spl-pinephone-SPEED.bin of=/dev/sdcard bs=1024 seek=8`

For eMMC installations, boot the device to [Jumpdrive](https://github.com/dreemurrs-embedded/Jumpdrive/releases) and follow the same procedure mentioned above to the eMMC device.

Danct12, 2020-09-17:
Have you tried 492MHz? It is the last resort if 552 doesn't work.

The red LED blinking is a sign of kernel panic, and the DRAM patch as mentioned have nothing to do with system crashes (it only allows 3GB PinePhones to boot.

mogwa1, 2020-09-17:
Yes, I've tried the 492Mhz.  With that one it now seems to work as long as I don't put in a data SD card (so not an SD card which is supposed to boot).  It gets up to the PIN entry and then runs into a kernel panic.
I'll see if I can somehow get a trace.

That same data SD card is working perfectly fine in the braveheart, and I can also mount it in other devices.  So it shouldn't be a problem with a corrupt card.

mogwa1, 2020-09-17:
Aha, luckily there is still a log of the kernel panic.

```Sep 14 14:08:15 alarm kernel: ------------[ cut here ]------------
Sep 14 14:08:15 alarm kernel: WARNING: CPU: 2 PID: 0 at kernel/rcu/tree.c:644 rcu_irq_exit+0x84/0x90
Sep 14 14:08:15 alarm kernel: Modules linked in: qmi_wwan cdc_wdm option usbnet usb_wwan usbserial mii rfcomm bnep hci_uart bt>
Sep 14 14:08:15 alarm kernel: CPU: 2 PID: 0 Comm: swapper/2 Tainted: G         C        5.7.19-4-danctnix #1
Sep 14 14:08:15 alarm kernel: Hardware name: Pine64 PinePhone (1.2) (DT)
Sep 14 14:08:15 alarm kernel: pstate: 60000085 (nZCv daIf -PAN -UAO)
Sep 14 14:08:15 alarm kernel: pc : rcu_irq_exit+0x84/0x90
Sep 14 14:08:15 alarm kernel: lr : irq_exit+0xd0/0xe0
Sep 14 14:08:15 alarm kernel: sp : ffff800010013ca0
Sep 14 14:08:15 alarm kernel: x29: ffff800010013ca0 x28: ffff0000abcca940 
Sep 14 14:08:15 alarm kernel: x27: 0000000000000030 x26: ffff800010fd60d0 
Sep 14 14:08:15 alarm kernel: x25: 00000000010f0000 x24: ffff0000abcca940 
Sep 14 14:08:15 alarm kernel: x23: ffff80001001f000 x22: ffff0000af9999c0 
Sep 14 14:08:15 alarm kernel: x21: ffff800010fdad68 x20: ffff800010fc3018 
Sep 14 14:08:15 alarm kernel: x19: ffff0000abcca940 x18: 0000000000000000 
Sep 14 14:08:15 alarm kernel: x17: 0000000000000000 x16: 0000000000000000 
Sep 14 14:08:15 alarm kernel: x15: 0000005b01d7f932 x14: 00000000000000ee 
Sep 14 14:08:15 alarm kernel: x13: 0000000000000001 x12: 0000000000000001 
Sep 14 14:08:15 alarm kernel: x11: ffff0000af9953b0 x10: 0000000000000001 
Sep 14 14:08:15 alarm kernel: x9 : 0000000000000000 x8 : ffff800010013e60 
Sep 14 14:08:15 alarm kernel: x7 : 0000000000000000 x6 : 000000002bcf09bc 
Sep 14 14:08:15 alarm kernel: x5 : 00ffffffffffffff x4 : 001312d00aa49c00 
Sep 14 14:08:15 alarm kernel: x3 : 0000000000000000 x2 : ffff800010013ca0 
Sep 14 14:08:15 alarm kernel: x1 : ffff0000af99a5c0 x0 : ffff800010fcf5c0 
Sep 14 14:08:15 alarm kernel: Call trace:
Sep 14 14:08:15 alarm kernel:  rcu_irq_exit+0x84/0x90
Sep 14 14:08:15 alarm kernel:  irq_exit+0xd0/0xe0
Sep 14 14:08:15 alarm kernel:  scheduler_ipi+0xac/0x1b0
Sep 14 14:08:15 alarm kernel:  handle_IPI+0xe4/0x178
Sep 14 14:08:15 alarm kernel:  gic_handle_irq+0xa4/0xa8
Sep 14 14:08:15 alarm kernel:  el1_irq+0xb8/0x140
Sep 14 14:08:15 alarm kernel:  sched_group_set_shares+0x62c/0x760
Sep 14 14:08:15 alarm kernel:  _stext+0x120/0x25c
Sep 14 14:08:15 alarm kernel:  irq_exit+0xc0/0xe0
Sep 14 14:08:15 alarm kernel:  __handle_domain_irq+0x68/0xc0
Sep 14 14:08:15 alarm kernel:  gic_handle_irq+0x54/0xa8
Sep 14 14:08:15 alarm kernel:  el1_irq+0xb8/0x140
Sep 14 14:08:15 alarm kernel:  arch_cpu_idle+0x14/0x20
Sep 14 14:08:15 alarm kernel:  do_idle+0x210/0x288
Sep 14 14:08:15 alarm kernel:  cpu_startup_entry+0x28/0x98
Sep 14 14:08:15 alarm kernel:  secondary_start_kernel+0x138/0x170
Sep 14 14:08:15 alarm kernel: ---[ end trace 3e87c8020b80a748 ]---
Sep 14 14:08:15 alarm kernel: Insufficient stack space to handle exception!
Sep 14 14:08:15 alarm kernel: ESR: 0x96000047 -- DABT (current EL)
Sep 14 14:08:16 alarm kernel: ------------[ cut here ]------------
Sep 14 14:08:16 alarm kernel: FAR: 0xffff80001000fff0
Sep 14 14:08:16 alarm kernel: WARNING: CPU: 0 PID: 0 at kernel/rcu/tree.c:569 rcu_idle_enter+0x74/0x80
Sep 14 14:08:16 alarm kernel: Task stack:     [0xffff8000110e0000..0xffff8000110e4000]
Sep 14 14:08:16 alarm kernel: Modules linked in: qmi_wwan cdc_wdm option usbnet usb_wwan usbserial mii rfcomm bnep hci_uart bt>
Sep 14 14:08:16 alarm kernel: IRQ stack:      [0xffff800010010000..0xffff800010014000]
Sep 14 14:08:16 alarm kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        WC        5.7.19-4-danctnix #1
Sep 14 14:08:16 alarm kernel: Overflow stack: [0xffff0000af98e090..0xffff0000af98f090]
Sep 14 14:08:16 alarm kernel: Hardware name: Pine64 PinePhone (1.2) (DT)
Sep 14 14:08:16 alarm kernel: CPU: 2 PID: 0 Comm: swapper/2 Tainted: G        WC        5.7.19-4-danctnix #1
Sep 14 14:08:16 alarm kernel: pstate: 80000085 (Nzcv daIf -PAN -UAO)
Sep 14 14:08:16 alarm kernel: Hardware name: Pine64 PinePhone (1.2) (DT)
Sep 14 14:08:16 alarm kernel: pc : rcu_idle_enter+0x74/0x80
Sep 14 14:08:16 alarm kernel: pstate: 000003c5 (nzcv DAIF -PAN -UAO)
Sep 14 14:08:16 alarm kernel: lr : do_idle+0x20c/0x288
Sep 14 14:08:16 alarm kernel: pc : rcu_nmi_enter+0x4/0x70
Sep 14 14:08:16 alarm kernel: sp : ffff800010fd3ec0
Sep 14 14:08:16 alarm kernel: lr : do_debug_exception+0x64/0x1a8
Sep 14 14:08:16 alarm kernel: x29: ffff800010fd3ec0 x28: 0000000040f30018 
Sep 14 14:08:16 alarm kernel: sp : ffff800010010000
Sep 14 14:08:16 alarm kernel: x27: 00000000fdf6cd30 x26: 0000000000000000 
Sep 14 14:08:16 alarm kernel: x29: ffff800010010000 x28: ffff0000abcca940 
Sep 14 14:08:16 alarm kernel: x25: 0000000000000000 x24: ffff800010fdb03c 
Sep 14 14:08:16 alarm kernel: x27: 0000000000000000 x26: ffff800010014000 
[... 3 lines cut ...]
Sep 14 14:08:16 alarm kernel: x23: 00000000000003c5 x22: ffff8000101737a4
Sep 14 14:08:16 alarm kernel: x19: ffff800010fdafa0 x18: 00000000000002e1 
Sep 14 14:08:16 alarm kernel: x21: ffff8000101737a4 x20: 00000000f2000800 
Sep 14 14:08:16 alarm kernel: x17: 0000000000000001 x16: 0000000000000001 
Sep 14 14:08:16 alarm kernel: x19: ffff800010010050 x18: 0000000000000394 
Sep 14 14:08:16 alarm kernel: x15: 0000000000000000 x14: 00000000000002e1 
Sep 14 14:08:16 alarm kernel: x17: 0000000000000002 x16: 0000000000000003 
Sep 14 14:08:16 alarm kernel: x13: 0000000000000001 x12: 0000000000000000 
[... cut ...]
Sep 14 14:08:16 alarm kernel: x3 : 3ffffffffffffffc x2 : 4000000000000000 
Sep 14 14:08:16 alarm kernel: x5 : ffff0000af99a5c0 x4 : 0000aaaae5dde7f0 
Sep 14 14:08:16 alarm kernel: x1 : ffff800010fcf5c0 x0 : ffff0000af96e5c0 
Sep 14 14:08:16 alarm kernel: x3 : ffff80009e9cb000 x2 : 0000000000000000 
Sep 14 14:08:16 alarm kernel: Call trace:
Sep 14 14:08:16 alarm kernel: x1 : 00000000000003c5 x0 : 0000aaaae5dde7f0 
Sep 14 14:08:16 alarm kernel:  rcu_idle_enter+0x74/0x80
Sep 14 14:08:16 alarm kernel: Kernel panic - not syncing: kernel stack overflow```

Danct12, 2020-09-17:
Does this happen with postmarketOS as well?

mogwa1, 2020-09-17:
Yes, that's something I also wanted to check.

At least it didn't crash on the first boot with postmarketOS.  At that moment it didn't have an SD card or SIM card.
It did crash after 30 seconds on first boot after flashing with Arch (also without SD card or SIM card).
Of course the difference could be that postmarketOS is using a different default DRAM speed setting.

I'll try it out with postmarketos and the full setup with SD card and SIM card, and let you know.
Thanks for the help so far.

(PS: In case that it doesn't immediately crash, I'm also hearing a couple of very loud clicking sounds around the time the 4G and wifi are connecting.  Again, the braveheart is not doing that.)

yaymuffins, 2020-09-17:
Just to offer some opposite feedback, but on a new 3GB Pinephone the distro work fine here? @mogwa1 probably lost the silicon lottery :<

mogwa1, 2020-09-17:
Yes, looks like I've lost the silicon lottery somehow.  Firstly, the DRAM doesn't seem to be particularly good on this device.  And then there's the hard kernel panic issue...

I've just tested with a fresh pmOS install.  It's booting ok until I insert the data SD card.  In that case it already gets stuck at the point where the LEDs are lighting up right after the device starts booting.  If I remove the card and do a hard reset, it passes that same point within 2 seconds or so.
So it's crashing even faster with pmOS.  Very strange.

I'll close this issue now since it doesn't seem to be distro-related after all.
Thanks again for offering your help to debug.

closed 2020-09-17 by mogwa1

Danct12, 2020-09-17:
That is not great, I hope they'll accept the RMA. You can email them via support@pine64.org.

Danct12, 2020-09-28:
@mogwa1, anything new regarding the device?

mogwa1, 2020-09-28:
Hi @Danct12, thanks for asking.
I've spent some more time debugging over the last couple of days, and I think I've got the issues pinned down now.  The good news is that the device has now been operating stably for several days!

I turns out that the hard kernel panics were due to a u-boot loader still being present on the SD card.  I had used that card to test out some distros before, and it was stupid of me to not wipe the u-boot sectors on the card before repartitioning and using it as data card.  The weird thing is that the braveheart device does not seem to mind that the u-boot loader was present on the card.  What makes it even more strange is that the 3gb device would sometimes still boot, but then crash shortly after.  I still don't know why that happened, but those crashes have completely stopped after wiping the relevant sectors on the SD card.

The "soft" crashes that I had seem to be completely related to the DRAM speed.  They haven't happened since I've flashed the lowest DRAM speed u-boot version.

In the end, it was this combination of two different types of crashes that complicated the debugging.  Both of those crashes were happening during boot or directly after boot, which made it difficult to get the logs and find the root cause.

Anyway, it seems that the device itself is now operating nicely (although at low DRAM speed).  Thanks again for the help and the hard work that you put into getting Arch to run on the pinephone!

Danct12, 2020-09-28:
Such a happy ending then. :)