dreemurrs-embedded/Pine64-Arch #61: Gnome-Image fir PineTab

rakor, 2020-09-18:
Will you bring a Gnome-Image for the PineTab? Would be great as Gnome has decent touch-support, us a full working desktop and should run fine on the screensize and resolution.

Danct12, 2020-09-19:
This is something we'll think of, but GNOME is kinda slow on the Mali 400.

pitlochry, 2020-10-02:
What do I have to install/remove and enable/disable in order to use gnome-shell and not phosh/phoc?

I think that gnome without phosh/phoc should be usable on a screen like the pinetab (at least on the surface tablets gnome is totally usable).

Danct12, 2020-10-05:
`sudo systemctl disable phosh` should be enough to stop phosh from starting.

Though, if you want to uninstall Phosh then run `pacman -Rscn danctnix-phosh-ui-meta`.

Danct12, 2020-12-22:
Duplicate of https://github.com/dreemurrs-embedded/Pine64-Arch/issues/34

closed 2020-12-22 by Danct12