dreemurrs-embedded/Pine64-Arch #62: After update of systemd, "squeekboard" is not auto-started

cookiengineer, 2020-09-18:
As the title says, I updated via `pacman -Syu` and luckily had openssh installed and enabled as a systemd service before it happened.

An update of systemd will cause `squeekboard` to be not auto-started, I'm not sure why exactly this is the case.
It seems as the `/etc/systemd/system/graphical.target.wants/phosh.conf` service is not enough.

How to reproduce:

```
sudo pacman -Syu;
sudo reboot;

# wait for boot, then login via ssh from another Computer
squeekboard &; # will make keyboard activateable again via the button on the bottom left in phosh
```

cookiengineer, 2020-09-18:
Additionally, there's a coredump generated by systemd for every reboot. They look all like this:

```bash
[alarm@alarm system]$ coredumpctl info 4158
           PID: 4158 (squeekboard)
           UID: 1000 (alarm)
           GID: 1000 (alarm)
        Signal: 5 (TRAP)
     Timestamp: Fri 2020-09-18 19:52:48 UTC (46min ago)
  Command Line: squeekboard --help
    Executable: /usr/bin/squeekboard
 Control Group: /user.slice/user-1000.slice/session-c2.scope
          Unit: session-c2.scope
         Slice: user-1000.slice
       Session: c2
     Owner UID: 1000 (alarm)
       Boot ID: 0b1ecf0ed25041bb954f229755518371
    Machine ID: 9b8685e54b0047dc89eb41043dd14875
      Hostname: alarm
       Storage: /var/lib/systemd/coredump/core.squeekboard.1000.0b1ecf0ed25041bb954f229755518371.4158.1600458768000000.zst
       Message: Process 4158 (squeekboard) of user 1000 dumped core.
                
                Stack trace of thread 4158:
                #0  0x0000ffffab2999d8 raise (libpthread.so.0 + 0x139d8)
                #1  0x0000ffffabf5e7d0 g_log_default_handler (libglib-2.0.so.0 + 0x647d0)
                #2  0x0000ffffabf5ea54 g_logv (libglib-2.0.so.0 + 0x64a54)
                #3  0x0000ffffabf5ec78 g_log (libglib-2.0.so.0 + 0x64c78)
                #4  0x0000aaaadb06452c n/a (squeekboard + 0xcb52c)
                #5  0x0000ffffac215fd4 n/a (libgio-2.0.so.0 + 0x14ffd4)
                #6  0x0000ffffac216340 n/a (libgio-2.0.so.0 + 0x150340)
                #7  0x0000ffffac19a500 n/a (libgio-2.0.so.0 + 0xd4500)
                #8  0x0000ffffac19b4a0 n/a (libgio-2.0.so.0 + 0xd54a0)
                #9  0x0000ffffac20a3d8 n/a (libgio-2.0.so.0 + 0x1443d8)
                #10 0x0000ffffac19a500 n/a (libgio-2.0.so.0 + 0xd4500)
                #11 0x0000ffffac19a554 n/a (libgio-2.0.so.0 + 0xd4554)
                #12 0x0000ffffabf5664c g_main_context_dispatch (libglib-2.0.so.0 + 0x5c64c)
                #13 0x0000ffffabf56ad8 n/a (libglib-2.0.so.0 + 0x5cad8)
                #14 0x0000ffffabf56ee0 g_main_loop_run (libglib-2.0.so.0 + 0x5cee0)
                #15 0x0000aaaadb0641ac n/a (squeekboard + 0xcb1ac)
                #16 0x0000ffffab111538 __libc_start_main (libc.so.6 + 0x24538)
                #17 0x0000aaaadb064408 n/a (squeekboard + 0xcb408)
                #18 0x0000aaaadb064408 n/a (squeekboard + 0xcb408)
```

cookiengineer, 2020-09-18:
A systemd service file for squeekboard seems to fix this issue. I'm not sure whether it's actually phosh's responsibility to make sure squeekboard is restarted or not, but this user service file worked for me:

**1: Save as /usr/lib/systemd/user/squeekboard.service**

```systemd
[Unit]
Description=Squeekboard, a mobile keyboard
Documentation=https://source.puri.sm/Librem5/squeekboard

[Service]
ExecStart=/usr/bin/squeekboard
TimeoutStartSec=10
Restart=on-failure
RestartSec=5s

[Install]
WantedBy=default.target
```

**Enable squeekboard.service**

```bash
systemctl --user enable squeekboard;
```

Danct12, 2020-09-27:
I cannot reproduce this on my device running 2020/09/13.

Danct12, 2020-10-03:
I'll close this for now, please reopen if the issue persists with the new image.

closed 2020-10-03 by Danct12