dreemurrs-embedded/Pine64-Arch #64: MAC address is randomly changing

Danct12, 2020-09-20:
With the latest reflash, it seems like MAC address keeps changing on each reboot.

edgar-vincent, 2020-09-21:
From your comment it seems that it wasn't the case with previous images. I don't know what caused this change, but this behaviour stops when NetworkManager is configured with the following settings:

```
# /etc/NetworkManager/conf.d/wifi_rand_mac.conf

[device-mac-randomization]
# "yes" is already the default for scanning
wifi.scan-rand-mac-address=yes
 
[connection-mac-randomization]
# Generate a random MAC for each WiFi and associate the two permanently.
wifi.cloned-mac-address=stable
```

dennis1248, 2020-09-26:
This might actually be desired behavior for a mobile phone. The device constantly changing MAC address will prevent it from being tracked using the MAC address. It's a common thing some Android devices do for this exact reason.

I am of the opinion however that it's better to not have it switch MAC randomly, if someone wants this functionality they can enable it themselves. Better to keep the MAC static to prevent people from running in to issues with DHCP and other applications.

Danct12, 2020-09-27:
Fixed, soon will be made to the new release.

Danct12, 2020-09-27:
Fixed, will soon make it to the next release.

closed 2020-09-27 by Danct12

InFerYes, 2020-11-01:
I disagree with @dennis1248 's sentiment. Since it's already done this way on Android devices, which is a lot of devices, I can assume that people are already dealing with DHCP issues. It would be a larger outcry if random MACs was really a major issue for non-tech people. I suggest leaving it to randomize for anti-tracking purposes and let people who really want static MACs to enable it themselves. This will make the OS more privacy oriented ootb. @Danct12