dreemurrs-embedded/Pine64-Arch #65: libpurple carbons

SevereOverfl0w, 2020-09-22:
I'm using the xmpp client in chats, and it would be useful to have carbons.  They support it by default, and there's already an AUR package: https://aur.archlinux.org/packages/libpurple-carbons-git/

(Haven't installed it yet myself, but I'll try to report back how I get on)

SevereOverfl0w, 2020-09-22:
Seems to work, I see the option tickbox in chatty.

genofire, 2020-11-09:
I use dino with some changes:

```
# Maintainer: Seppia <seppia@seppio.fish>
# Contributor: svalo <me@valo.space>

pkgname=dino-git
pkgver=v0.1.0.r152.g30b850c
pkgrel=1
pkgdesc="Simple and modern Jabber/XMPP client written in vala"
arch=('i686' 'x86_64' 'aarch64')
url="https://dino.im/"
license=('GPL3')
depends=('glib2>=2.38' 'glib-networking' 'gtk3-mobile>=3.22' 'gpgme' 'libgee>=0.10' 'libgcrypt' 'libsoup' 'sqlite' 'qrencode' 'libsignal-protocol-c' 'libhandy')
makedepends=('git' 'cmake' 'vala>=0.34' 'ninja')
optdepends=('libcanberra: audio notification support')
provides=("${pkgname%-git}")
conflicts=("${pkgname%-git}")
source=('dino::git+https://github.com/mimi89999/dino#branch=handy')
sha512sums=('SKIP')

### Any or all of the following plugin can be disabled: omemo, openpgp, http-files. Populate following string as per your needs and uncomment it.

# _DISABLE_PLUGINS="--disable-plugin='omemo;openpgp;http-files'"

### Any or all of the following plugin can be enabled: notification-sound. Populate the following string as per you needs and uncomment it.
### Install any needed optional dependency before

#_ENABLE_PLUGINS="--enable-plugin='notification-sound'"

### Since 55b148a0f78c1abef57c21b2a84d97cad32b2290 dino uses by default the shared libsignal library.
# To restore the old behaviour add '--with-libsignal-in-tree' flag to configure

pkgver() {
    cd "${srcdir}/${pkgname%-git}"
    git describe --long --tags | sed 's/\([^-]*-g\)/r\1/;s/-/./g'
}

build() {
    cd "${srcdir}/${pkgname%-git}"
    ./configure $_DISABLE_PLUGINS $_ENABLE_PLUGINS --with-tests --prefix="/usr" CC="$CC" CFLAGS="$CFLAGS" VALACFLAGS="$VALACFLAGS"
    make
}

package() {
    cd "${srcdir}/${pkgname%-git}"
    make DESTDIR="${pkgdir}/" install
}

check() {
    cd "${srcdir}/${pkgname%-git}"
    echo "Executing xmpp-vala-test:"
    build/xmpp-vala-test
}
```

commit d5c61494 (2021-01-21): pkg: danctnix: libpurple-carbons: new pkgbuild

Closes #65 #117
closed 2021-01-21 by Danct12