dreemurrs-embedded/Pine64-Arch #67: Missing kernel modules

daniellandau, 2020-09-29:
I'm especially interested in `CONFIG_CAN`, `CONFIG_CAN_RAW` and `CONFIG_CAN_GS_USB`, but I'd imagine there's a bunch of other modules for various USB stuff that might benefit from modules being built.

genofire, 2020-11-07:
Here are much more, for running `iwd` (wpa_supplicated was already replaced with iwd on archlinux iso ...):
```
Nov 07 18:17:15 alarm iwd[3491]: No HMAC(SHA1) support found
Nov 07 18:17:15 alarm iwd[3491]: No HMAC(MD5) support found
Nov 07 18:17:15 alarm iwd[3491]: No CMAC(AES) support found
Nov 07 18:17:15 alarm iwd[3491]: No HMAC(SHA256) support not found
Nov 07 18:17:15 alarm iwd[3491]: No HMAC(SHA512) support found, certain TLS connections might fail
Nov 07 18:17:15 alarm iwd[3491]: DES support not found
Nov 07 18:17:15 alarm iwd[3491]: AES support not found
Nov 07 18:17:15 alarm iwd[3491]: No CBC(DES3_EDE) support found, certain TLS connections might fail
Nov 07 18:17:15 alarm iwd[3491]: No CBC(AES) support found, WPS will not be available
Nov 07 18:17:15 alarm iwd[3491]: No Diffie-Hellman support found, WPS will not be available
Nov 07 18:17:15 alarm iwd[3491]: The following options are missing in the kernel:
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_USER_API_HASH
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_USER_API_SKCIPHER
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_KEY_DH_OPERATIONS
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_ECB
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_MD5
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_CBC
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_SHA256
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_AES
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_DES
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_CMAC
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_HMAC
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_SHA512
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_SHA1
Nov 07 18:17:15 alarm iwd[3491]: The following optimized implementations might be available:
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_SHA1_SSSE3
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_AES_NI_INTEL
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_SHA512_SSSE3
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_AES_X86_64
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_DES3_EDE_X86_64
Nov 07 18:17:15 alarm iwd[3491]:         CONFIG_CRYPTO_SHA256_SSSE3
```

beckerichmika, 2020-11-16:
CONFIG_CRYPTO_USER is necessary for bluez, because it expects 'crypto_user' in its modules-load.d/bluez.conf file

beckerichmika, 2020-11-19:
We also need CONFIG_AUDIT to be enabled to get rid of 'audit: failed to open auditd socket: Protocol not supported' thrown by NetworkManager

NoahAndrews, 2020-12-16:
Are you thinking about putting a PinePhone on an FRC robot?

Danct12, 2021-04-02:
CONFIG_AUDIT has made it's way to 5.10.19-3.

closed 2021-04-02 by Danct12