dreemurrs-embedded/Pine64-Arch #69: phosh is not able to slide between open apps

philmmanjaro, 2020-10-04:
I just had tested 2020/09/27 build and noticed that within **phosh** you can't slide between the apps anymore. Maybe it is due to the auto-rotate patch or something else. With **mobian** nightly from today I was able to slide between apps there. However they don't ship the auto-rotate MR yet.

Pacman99, 2020-10-09:
Mobian used to have this same issue I think 3 days ago, so its most likely not because of the auto-rotate patch.. I think there was an update of phosh itself that fixed it, but I never debugged it fully because it just started working after the recent update.

philmmanjaro, 2020-10-11:
Well, it is fixed in 0.4.5 release of Phosh, however **auto-rotate** patch doesn't work anymore. It needs a rebase.

ikidd, 2020-10-12:
Is this auto-rotate patch related to how it's switched from locked/unlocked to portrait/landscape?  I can't say I care for that change, especially as it rotates landscape opposite to how I've ever seen it rotated.

Danct12, 2020-10-13:
Fixed in the latest update.

closed 2020-10-13 by Danct12

philmmanjaro, 2020-10-13:
Well, on my end it is only partly fixed, as we have to get between the apps into the spaces, to be able to slide. Have to check with latest updates on Arch image. @Danct12 if there was a setting or patch, please point me to it.