dreemurrs-embedded/Pine64-Arch #70: gedit crashes when closing without saving

grelltrier, 2020-10-08:
When you type any text into gedit and close it without saving all kinds of weird things will happen.
Usually the phone becomes unresponsive for a few seconds and there is a gap at the top between the "app drawer" and the top bar. Then one of two things will happen:
A: The phone gets a "blue screen" saying something has gone wrong and you need to log out. When pressing the "Log out" button, the phone shuts down but gets stuck so you need to press the power button to restart it
B: The phone displays the login and goes black. When pressing the power button, the login shows again. Sometimes everything works fine after that and sometimes the phone behaves completely unexpected. This goes from needing to press twice to activate buttons to not responding or doing even weirder things.

When I asked in the Pinephone group if anybody else notices it, tor.sh said it does not happen when using the flatpack version. Others using the version that comes with the distribution noticed the same behavior.

I think this issue came up somewhere around the release of Arch 20200913. I did not report this earlier because I thought the keyboard application I was developing caused the crashes -,-

grelltrier, 2020-10-09:
I guess this is more of a phosh bug:
https://source.puri.sm/Librem5/phosh/-/issues/384

Danct12, 2020-10-13:
Fixed in the latest update.

closed 2020-10-13 by Danct12