dreemurrs-embedded/Pine64-Arch #75: Can't wake up from suspend

Pacman99, 2020-10-18:
I can no longer wake up the phone by clicking the power button. This seems to have started since an update I ran on 10/13. 

I found this error in journalctl:
`/usr/lib/systemd/system-sleep/pinephone-suspend-hook.sh failed with exit status 1.`

Danct12, 2020-10-19:
Can you tell me which device are you on? BH or CE?

Pacman99, 2020-10-19:
I'm on the ubuntu CE

Pacman99, 2020-10-20:
Okay so I enabled that option "IT_NEVER_CAME_BACK_ON" the fix recommended for braveheart users. And its working perfectly now. 
Weird that it's happening now for an ubuntu CE pinephone. I guest the linux 5.9 update caused it, since thats the only thing that changed.

closed 2020-10-20 by Pacman99

Pacman99, 2020-10-23:
I flashed the 10/15 image and updated on that. I don't have the issue anymore and I didn't have to enable the fix this time.

ciol, 2020-10-28:
I've been having this issue as well, 3 devices, 3 different behaviours :
- on pinetab with latest updates it works without any changes required
- on pinephone Ubports CE with latest updates/flashed image
  Fixed by uncommenting `IT_NEVER_CAME_BACK_ON=1` in `/usr/lib/systemd/system-sleep/pinephone-suspend-hook.sh`
- on pinephone pmOS CE 3Gb version with latest updates/flashed image
  Tried uncommenting `IT_NEVER_CAME_BACK_ON` but that doesn't seem to work

I can also say that this happens only when the device is running off the battery, when it is plugged in it has no problem waking up from sleep.

I'm unsure what I can do next to try and fix this.