dreemurrs-embedded/Pine64-Arch #82: odd vibration

ArchNemSyS, 2020-10-28:
Following a fresh install of Arch Linux ARM - 2020/10/24 to pinephone (3Gb version) eMMC 

The phone will start a continuous vibration sometime after boot this appears to happen when idle and in use whilst on charge and not, it does not however occur in deep sleep.

No notification is given, Phosh is responsive there is no LED activity being unable to stop said vibration I then attempt to soft shutdown the device via the GUI at which point it freezes and hard off is the only option.

upon reboot journalctl doesn't reveal anything telling. 

I have tried lowering the thermal trip point and setting the CPU frequency to min and can confirm the bug still occurs at cool temps so I don't think its thermal.     

I am using zramswap and the default apps only in a similar configuration to an install [2020/10/15] I had on an SD Card that didn't experience this bug.

Is their any hardware scan/benchmark or logging I could perform to help identify the bug ?

ArchNemSyS, 2020-10-28:
Upon a subsequent reboot I have a red flashing LED upon opening the epiphany browser, so I'm guess kernel panic

Danct12, 2020-10-28:
If you're having kernel panic, please flash alternative U-Boot:

```
dd if=/boot/u-boot-sunxi-with-spl-pinephone-552.bin of=/dev/mmcblkX bs=1024 seek=8
```

Replace mmcblkX with the following:
- mmcblk0 (SD)
- mmcblk2 (eMMC)

ArchNemSyS, 2020-10-28:
Thankyou !!!

Currently streaming audio without issue

closed 2020-10-28 by ArchNemSyS

Danct12, 2020-10-28:
I'm glad you're also happy as well!