dreemurrs-embedded/Pine64-Arch #83: mobile-config-firefox requires firefox 79

willjp, 2020-11-10:
upstream postmarketOS's `mobile-config-firefox` requires firefox-79 at the latest.
https://gitlab.com/postmarketOS/mobile-config-firefox/

I'll try making a PR when I have some extra time. In the meantime here's a workaround:

### workaround

I was able to workaround by installing the package by hand:
```bash
# download/install
curl -#0 http://tardis.tiny-vps.com/aarm/packages/f/firefox/firefox-79.0-1-aarch64.pkg.tar.xz
sudo pacman -U firefox-79.0-1-aarch64.pkg.tar.xz
```

Then I'm avoiding future conflicts on update by pinning the package
```conf
# /etc/pacman.conf

IgnorePkg = firefox mobile-config-firefox
```

Danct12, 2020-11-14:
The config still works perfectly fine with FF 82, so don't think this is currently a issue now.

willjp, 2020-11-17:
Thanks! I haven't updated yet - I'll close this.
(Also, thank you so much for all of the work you've put into this!)

closed 2020-11-17 by willjp