dreemurrs-embedded/Pine64-Arch #85: Kernel panic on PinePhone v1.2b with 20201024

tobymurray, 2020-11-11:
Hardware: PinePhone v1.2b Convergence package (3GB)
Version: https://github.com/dreemurrs-embedded/Pine64-Arch/releases/tag/20201024
First of all, I apologize for the horrifying screenshot of kernel panic but I'm not sure how to get the information off the PinePhone in a usable fashion.

[image]

I was using this image on my SD card with varying success, but it seemed to hang every few minutes and need a reboot. I just flashed 20201024 to the eMMC with Etcher and tried to boot but ended up with an immediate kernel panic.

Any thoughts on troubleshooting steps/how to get more information out (without resorting to pictures)?

Danct12, 2020-11-11:
Please flash an alternative U-Boot image (u-boot-sunxi-with-spl-pinephone-552.bin) by taking out the SD card or using Jumpdrive using this command:

`dd if=u-boot-sunxi-with-spl-pinephone-SPEED.bin of=/dev/mmcblk bs=1024 seek=8` (replace SPEED with 552 first, if that still crashes you can try 492)

tobymurray, 2020-11-12:
Previously what I did was use Etcher on Windows and flash the SD card with archlinux-pinephone-20201024.img.xz, then I flashed Jumpdrive and flashed the eMMC.

I moved to my Ubuntu machine to use dd, but my first attempt has not shown a kernel panic. I'll try doing everything as much the same and I'll see how it goes, if I can reproduce the kernel panic.

Lin-Admin, 2020-11-14:
With alarm 5.9.3-1-danctnix #1 SMP Sat Nov 7 this seems not to happen.

Danct12, 2020-11-15:
Personally, I think that had to do with the new build using 552MHz by default for DRAM, previous builds uses 624 which is much faster, but seems not too stable on some devices.

closed 2020-11-15 by Danct12