dreemurrs-embedded/Pine64-Arch #88: dmesg shows many warnings and errors

Lin-Admin, 2020-11-14:
With "alarm 5.9.3-1-danctnix #1 SMP Sat Nov 7"  dmesg shows many warnings and errors. IMHO it is worth the effort to eliminate some of them.

Danct12, 2020-11-14:
Please clarify on what these warnings/errors are.

philmmanjaro, 2020-11-15:
Seems the same question was asked on our bug tracker. Would be nice to see more details on which warnings or errors are meant here.

Lin-Admin, 2020-11-16:
Could it be that competent developers can not have a look at their dmesg?
`[root@alarm alarm]# dmesg -Hl warn,err 
[Okt25 19:02] sun50i-a64-r-pinctrl 1f02c00.pinctrl: supply vcc-pl not found, using dummy regulator 
[  +0,142453] SMCCC: SOC_ID: ARCH_SOC_ID(0) returned error: ffffffffffffffff 
[  +0,040349] axp20x-adc axp813-adc: DMA mask not set 
[  +0,002766] axp20x-battery-power-supply axp20x-battery-power-supply: DMA mask not set 
[  +0,008497] axp20x-usb-power-supply axp20x-usb-power-supply: DMA mask not set 
[  +0,148844] usb_phy_generic usb_phy_generic.3.auto: supply vcc not found, using dummy regulator 
[  +0,001155] phy id_det change to device 
[  +0,114537] Goodix-TS 1-005d: Direct firmware load for goodix_917S_cfg.bin failed with error -2 
[  +0,007538] sun50i-a64-pinctrl 1c20800.pinctrl: supply vcc-pe not found, using dummy regulator 
[  +0,384731] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2 
[  +3,562222] dw-apb-uart 1c28000.serial: forbid DMA for kernel console 
[  +0,394061] systemd[1]: system-getty.slice: unit configures an IP firewall, but the local system does not support BPF/cgroup firewalling. 
[  +0,000012] systemd[1]: (This warning is only shown for the first unit using IP firewalling.) 
[  +3,413008] 8723cs: module is from the staging directory, the quality is unknown, you have been warned. 
[  +3,144875] bt-sco bt-sco-codec: ASoC: sink widget Capture overwritten 
[  +0,000015] bt-sco bt-sco-codec: ASoC: source widget Playback overwritten 
[  +0,000125] simple-amplifier audio-amplifier: supply VCC not found, using dummy regulator 
[  +0,010919] debugfs: File 'Playback' in directory 'dapm' already present! 
[  +0,000014] debugfs: File 'Capture' in directory 'dapm' already present! 
[  +0,497233] dw-apb-uart 1c28400.serial: failed to request DMA 
[  +0,371806] dw-apb-uart 1c28c00.serial: failed to request DMA 
[  +0,386833] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723cs_xx_config-pinephone.bin failed with error -2 
[  +0,000006] Bluetooth: hci0: RTL: mandatory config file rtl_bt/rtl8723cs_xx_config not found 
[ +13,730955] modem-power serial1-0: Your modem has an outdated firmware. Latest know version is EG25GGBR07A08M2G_01.002.07. Consider updating. 
[  +1,201859] modem-power serial1-0: AT command 'AT+QCFG="fast/poweroff"' returned ERROR 
`

beckerichmika, 2020-11-16:
The axp20x errors need dtb changes I think, which needs time. The VCC errors could be fixed if we put the drivers into the initramfs according to some research. The firmware errors could be indicating missing firmware.

Danct12, 2021-01-14:
Those errors aren't deemed to be critical and I suggest contacting megous at megous com if you're too aware about them. However:

- Touchscreen firmware is not needed, it is for firmware upgrade.
- Bluetooth firmware error is something you will get if you remove the rtl8723bs-firmware. (ERRNO -2 is "No such file/dir")
- Outdated firmware can be upgraded by following [this guide](https://github.com/dreemurrs-embedded/Pine64-Arch/wiki/Upgrade-the-modem-firmware)

closed 2021-01-14 by Danct12

Lin-Admin, 2021-01-16:
"Those errors aren't deemed to be critical "
That is not a valid reason to ignore the messages and the bug is still present!

Danct12, 2021-01-17:
> and the bug is still present!

You did not clarify on which bug you're getting.

megous, 2021-01-17:
@Lin-Admin 
- axp803 doesn't have/use DMA, so DMA mask warnings are superfluous, just ignore them
- dw-apb-uart DMA should not be requested because it's not configured in DT, so that's just another superfluous warning
- SMCCC: SOC_ID: ARCH_SOC_ID(0) returned error... was probably just useless warning in 5.10, it's gone in 5.11
- debugfs/ASoC warnings are curious. I tried fixing those a few months ago, but didn't find the root cause, but this time there's extra information about the source of the issue in dmesg, so it might be worth looking into those again.

megous, 2021-01-17:
@beckerichmika 
There are no VCC errors. Audio PA is supplied from PS (power switch), which is not modeled in DT, and VCC-PE is intentionally not configured to allow cameras' power supply to power down when not used.