dreemurrs-embedded/Pine64-Arch #90: Sound of mpv often stuttering

Lin-Admin, 2020-11-14:
With "alarm 5.9.3-1-danctnix #1 SMP Sat Nov 7", when playing a sound file with mpv, the output on the built-in jack is often stuttering.
This is even when no other app is running and cpu usage only a few pct.
Whenever another app is run, it becomes much worse.

lajonss, 2020-11-14:
I'm experiencing similar issue with BT audio and built-in speaker. Strawberry just stops the playback with message
```
17:37:59.064 INFO  GstEngine:940                    Got stream info for "file:///home/alarm/sd/tego/turbo/turbo/bonapetit.mp3:" "MPEG-1 Layer 3 (MP3)"                                        
17:38:03.322 ERROR GstEnginePipeline:852            ErrorMessageReceived ID: 1 Domain: 3193 Code: 1 Error: "Disconnected: Connection terminated"                                              
17:38:03.322 ERROR GstEnginePipeline:853            ErrorMessageReceived ID: 1 Domain: 3193 Code: 1 Debug: "../gst-plugins-good/ext/pulse/pulsesink.c(412): gst_pulsering_is_dead (): /GstPlayBin:playbin-0/GstPlaySink:playsink/GstBin:abin/GstBin:audiobin/GstAutoAudioSink:autoaudiosink-1/GstPulseSink:autoaudiosink-1-actual-sink-pulse"                                               
17:38:03.335 ERROR GstEnginePipeline:852            ErrorMessageReceived ID: 1 Domain: 3193 Code: 1 Error: "pa_stream_writable_size() failed: Connection terminated"                          
17:38:03.336 ERROR GstEnginePipeline:853            ErrorMessageReceived ID: 1 Domain: 3193 Code: 1 Debug: "../gst-plugins-good/ext/pulse/pulsesink.c(1750): gst_pulseringbuffer_commit (): /GstPlayBin:playbin-0/GstPlaySink:playsink/GstBin:abin/GstBin:audiobin/GstAutoAudioSink:autoaudiosink-1/GstPulseSink:autoaudiosink-1-actual-sink-pulse"                                         
17:38:03.343 ERROR GstEngine:563                    GStreamer error: 3193 1 "Disconnected: Connection terminated"                                                                             
17:38:03.354 ERROR logging:72(GLib)                 Source ID 10 was not found when attempting to remove it                                                                                   
17:38:03.415 WARN  unknown                          Wayland does not support QWindow::requestActivate()
```
It seems to affect only certain files though. So far I've been able to pin it down to 44.1kHz sampling rate - if the file is encoded with 48kHz the problem doesn't occur.

As a workaround you can try
```
ffmpeg -i <original file> -ar 48000 <output file>
```

lajonss, 2020-11-20:
Some logs from mpv trial:

Journalctl logs
```
lis 20 04:14:13 alarm bluetoothd[3418]: Endpoint unregistered: sender=:1.193 path=/MediaEndpoint/A2DPSink/sbc
lis 20 04:14:13 alarm bluetoothd[3418]: Endpoint unregistered: sender=:1.193 path=/MediaEndpoint/A2DPSource/sbc
lis 20 04:14:13 alarm systemd[3404]: pulseaudio.service: Main process exited, code=killed, status=9/KILL
```

pulseaudio -v
```
I: [alsa-sink-1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0] alsa-sink.c: Underrun!
I: [alsa-sink-1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0] alsa-sink.c: Increasing wakeup watermark to 30,00 ms
I: [alsa-sink-1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0] alsa-sink.c: Underrun!
I: [alsa-sink-1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0] alsa-sink.c: Increasing wakeup watermark to 40,00 ms
I: [alsa-sink-1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0] alsa-sink.c: Underrun!
I: [alsa-sink-1c22c00.dai-sun8i-codec-aif1 sun8i-codec-aif1-0] alsa-sink.c: Increasing wakeup watermark to 50,00 ms
Killed
```
[pulseaudio-vvvv.log](https://github.com/dreemurrs-embedded/Pine64-Arch/files/5571077/pulseaudio-vvvv.log)

ffprobe encoding data
```
  Duration: 00:04:29.79, start: 0.025057, bitrate: 131 kb/s
    Stream #0:0: Audio: mp3, 44100 Hz, stereo, fltp, 131 kb/s
    Metadata:
      encoder         : Lavc58.18
```

lajonss, 2020-11-25:
I've applied the pulseaudio update (and the .pacnew configuration).
So far I've been unable to kill pulseaudio with the same files. Tested with jack and BT headphones.

Danct12, 2021-01-14:
Fixed in the latest update.

closed 2021-01-14 by Danct12