dreemurrs-embedded/Pine64-Arch #91: Application "Software" crashes when i tap on any category

abysso2, 2020-11-16:
Hia ll,
when i start the application "Software" in order to search or install any softwarepackage, the application crashes when i e.g. tap on the category "Games" or "Audio & Video".

Does anyone here has the same issue?

Regards,
A.

wailord284, 2020-11-16:
I can confirm the same problem running the archlinux-pinephone-20201112.img.xz image on EMMC. I have done a full system upgrade as well and the problem persists. When pressing any category, the software app crashes and returns to the launcher. Upon relaunching the software app, it'll sometimes say "Software catalog is being downloaded". Even after this message though, pressing any of the category buttons causes a crash and the process repeats.
I can test more if anyone needs it.
Thanks.

abysso2, 2020-11-18:
Solved by the latest package update .... Thx a lot.

closed 2020-11-18 by Danct12

abysso2, 2020-11-20:
Hey Danct12,
how did you solve this issue? The guys from mobian got the same issue, so it would be cool to share youre wisdom :-)

Regards,
A.