dreemurrs-embedded/Pine64-Arch #98: Does Firefox an other apps fit the PinePhone-Screen?

abysso2, 2020-11-19:
Hi there,
dumb question: Does firefox, threema and other apps fit your screen? on my pinephone, the apps "are cut" on the right edge. Sorry for my poor english ...

Regards,
A.

aligator, 2020-11-19:
For Firefox there is a special package named mobile-config-firefox in the repo.
Some other apps work, some don't. Chromium for example works mostly, some apps (especially some gnome apps used in Phosh) are preinstalled in a mobile version.

But don't expect all programs to work fine.

If you manage to connect a HDMI screen (which is a bit buggy currently), you can drag the windows there and use them like on normal PCs.

abysso2, 2020-11-20:
Thanks a lot ...

closed 2020-11-20 by Danct12