dreemurrs-embedded/Pine64-Arch #100: Alarm schedule for several days does not work

kapinta, 2020-11-20:
Hello all,
If I set a single alarm in clocks app it works fine.
But if I try to schedule several alarms, for example, from monday to friday at 16h it does not work.
Could you please check this issue?
Thanks you!

asriel-danctnix, 2020-11-22:
The phone usually went to deep sleep when inactive, and there is no mechanism to make it wake up on clock yet, so it is adviced to keep the phone plugged to the charger (we disable sleep mode on charging).

kapinta, 2020-11-23:
Hello Asriel
I think that you should change again the tittle to the original one ;-)
I have done more test with the alarm. My pinephone is always enabled, I have disabled suspension feature always.
When I schedule an alarm for several days it even works fine, but if I restart the phone, the schedule is still visible, but the alarm does not sound.
I think that the problem is related to the alarm config after system restart. Maybe the alarm config should be read at boot time somewhere?
Thanks for the support.

kapinta, 2020-12-11:
Hello again!
After the last update between yesterday and today the alarm schedule works fine, even after rebooting the system.
I could not try it after system suspension.
Thank you!

kclisp, 2022-02-20:
@dariox86 For reference, Clocks should now autostart by [this patch](https://github.com/dreemurrs-embedded/Pine64-Arch/blob/master/PKGBUILDS/phosh/gnome-clocks-mobile/0011-autostart-gnome-clocks.patch).

Also for reference, upstream has [a merge request](https://gitlab.gnome.org/GNOME/gnome-clocks/-/merge_requests/165) and postmarketOS has [patched](https://gitlab.com/postmarketOS/pmaports/-/merge_requests/2342) it somewhat. PostmarketOS also has [a todo list](https://gitlab.com/postmarketOS/pmaports/-/issues/1170) regarding this issue. Unfortunately it doesn't seem like anyone is actively working on this. Maybe it'd be possible to port postmarketOS's patch to Arch?

dariox86, 2022-02-20:
I will test it again and report here.

Danct12, 2022-02-21:
Have you tried gnome-clocks-mobile?

dariox86, 2022-02-21:
I haven't. I have just installed gnome-clocks-mobile which replaces gnome-clocks. I will let you know.

dariox86, 2022-02-22:
As it has happened with other packages, gnome-clocks was not automatically replaced with gnome-clocks-mobile. Now it works! Thank you.