dreemurrs-embedded/Pine64-Arch #99: No notification created when receiving a new SMS

kapinta, 2020-11-20:
Hello all,
I have noticed that when I receive a new SMS on arch-pine it does not appear any kind of notification.
LED light does not work and no notification screen appears.
You can only see new SMSs after opening the app.
Thank you. Best regards.

InFerYes, 2020-11-23:
What app are you using for the sms? If Chatty, you can report it [here](https://source.puri.sm/Librem5/chatty/-/issues).

kapinta, 2020-11-23:
Hello @InFerYes 
I am using the default app for sms in arch, it is called chats, and it's release is 0.2.0.
Thanks.

Ocawesome101, 2021-02-25:
I've been having this issue too, for about a week, but *only after a suspend/resume cycle.*  My solution has been to disable suspend.  I get SMS after a reboot so I'd assume it's a Chatty or ModemManager issue.

Went ahead and opened an issue on the Chatty GitLab.

kapinta, 2021-08-11:
It seems that this issue was solved with an update because it is working fine now.
Thank you!

closed 2021-08-11 by kapinta