dreemurrs-embedded/Pine64-Arch #101: Questions: in which file is / are the access point/s configured

abysso2, 2020-11-22:
Hi,
i am connected to my pinephone via ssh and i am wondering, which file i have to edit to delete a old access point entry?

Any hints?

Regards,
A.

InFerYes, 2020-11-23:
How are the APs managed?

abysso2, 2020-11-23:
The AP ist automatically added after installation, entering the sim pin and activating "mobile data". Its name is "Prepaid Contracts" and an AP from t-mobile is configured correctly.

But after the first reboot, This entry "Prepaid Contracts" is doubled and i can not delete one of them ...

mogwa1, 2020-12-04:
It seems like it's related to the problem I had a while ago: #55 
I also got duplicate entries for the AP, and couldn't switch between APs from the settings menu.
I ended up manually editing the entry/entries in /etc/NetworkManager/system-connections

abysso2, 2020-12-04:
Jupp, this is my issue .. i will edit  /etc/NetworkManager/system-connections.

THANK YOU!

closed 2020-12-07 by Danct12