dreemurrs-embedded/Pine64-Arch #102: p64.arikawa-hi.me : SSL certificate problem: certificate has expired

grelltrier, 2020-11-23:
Updating packages via 'pacman -Syu' fails due to the expired certificate

Danct12, 2020-11-23:
The certificate is still valid for me, have you checked your time and date on the device?

grelltrier, 2020-11-23:
Is it not 15 January, 2116? :D
[image]

Never mind then. I never bothered setting the time since I only use the phone to test applications on it. Its weird it changed itself like that though. It updated like a charm for the last weeks and I didn't change any of the time settings. Automatically updating the time does not seem to work for me

grelltrier, 2020-11-25:
Apparently this has happened to people using Mobian too. According to the link the cause is still unknown but is likely to have something to do with the time synchronization.
https://wiki.mobian-project.org/doku.php?id=troubleshooting#year-2115-problem

Danct12, 2021-06-11:
This should be fixed with this patch: https://patchwork.kernel.org/project/linux-arm-kernel/patch/20210515021439.55316-1-samuel@sholland.org

Which is also a part of the megi kernel.

closed 2021-06-11 by Danct12