dreemurrs-embedded/Pine64-Arch #103: Numeric keyboard is only shown after setting up a new alarm

kapinta, 2020-11-23:
Hello all,
When setting a new alarm in clocks app, the numeric keyboard appears, but after closing this app, or going to another, the numeric keyboard keeps appearing and it cannot be selected any other kind of keyboard.
Thanks for the help.

Danct12, 2020-11-29:
Cannot reproduce, can you please let me how in steps?

kapinta, 2020-12-05:
Sure!
Try to set any alarm, and enable numeric keyboard to set the time you want to schedule the alarm by clicking hour or minute field, the numeric keyboard will appear, and after that save the alarm.
If you switch to any other app, for instance telegram or mail, it keeps appearing the numeric keyboard. I have not found the way to change to the standard keyboard.
As a workaround I have found that if you set the alarm, and maintain the focus on the alarm description field instead of the hour/minute fields, the problem does not appear.
Thanks for your support.

Danct12, 2020-12-07:
Cannot reproduce it, is it possible for you to make a video about it?

kapinta, 2020-12-11:
Video sent by telegram to Danct12

Danct12, 2020-12-12:
I have figured out how to reproduce the issue thanks to the video on Telegram.

I have opened a new report on upstream, it can be seen here: https://source.puri.sm/Librem5/squeekboard/-/issues/256

Danct12, 2020-12-16:
Fixed with 958f4d54e517f3a5110822825722a3f86c0dd9ce

closed 2020-12-16 by Danct12